frankcrawford/it87 #3: Mainlining the module delta?

iam-TJ, 2022-11-05:
Are you aware of any attempts, or impediments, to getting the delta between this and the mainline it87 module merged into main-line? It would obviously be beneficial to everyone.

If there are impediments (possibly documented in historic mailing lists) it'd be good to collect them here and see if we can address them so changes can eventually go directly into mainline.

goc9000, 2022-12-15:
I've been thinking about this as well... New versions of the ITx chips pop up seemingly every year, and each time the driver would need to be modified. And support for each individual chip will include guesswork for a long time, possibly forever, and the conservative kernel guys don't want to sign off on something so unstable. That's why this repo has all these hundreds of clones and why the original author quit in disgust.

One possible solution I've been considering relies on the fact that all these chips have a very similar structure (at least their environment controller part). There are usually only small changes from one to another, e.g. the number of fans, ADC resolution, presence or absence of certain features etc. It makes sense to extract a common model and just apply different parameters to it for each chip. The driver already does that, but the parameters and chip types are hardcoded. What if it exposed an interface (via the module config or maybe even sysfs entries) by which it could be configured for new chips from userspace? Kinda like force_id, except not just the ID but also the features, number of fans etc. That way we can unofficially experiment with new chips as soon as they are available, and maybe with time the configuration can be officialized and added to the driver once it's stood the test of time.

Oh and it also needs to support potentially different configurations for each address (there are already mobos with two different chips, and force_id doesn't work well for them).

frankcrawford, 2022-12-19:
@goc9000  that seems like an interesting idea and something we should try, although I'm probably not the best programmer to write it.  I'm willing to add in code for this, possibly on a separate branch for the moment.
On the other hand given a bit of time I will look at doing something about force_id for multiple chips, as I agree that it is currently a problem.

frankcrawford, 2023-04-16:
@goc9000 just a small update, things are progressing to get this into the mainline, and even some recent changes are in either 6.2 or will be in 6.3, such as the ability to use force_id for multiple different chips.

We are gradually getting there.

sergiotca, 2023-06-06:
> @goc9000 just a small update, things are progressing to get this into the mainline, and even some recent changes are in either 6.2 or will be in 6.3, such as the ability to use force_id for multiple different chips.
> 
> We are gradually getting there.

I cannot thank you enough for this buddy, keep up the great work!

iam-TJ, 2023-06-29:
@frankcrawford I'm interested in developing code based on @goc9000 idea of using a common model structure if we can thrash out a design that can be iterated on for testing to replace all the hard-coded specifications.

In addition it should have a userspace interface for adding/changing/removing chip specifications. That could be via sysfs but I suspect might be frowned upon by kernel devs since it'd need to be exposing a lot of internals to do it justice, so I wonder about providing a netlink interface and a companion userspace tool `it87ctl` to use the netlink interface for the required operations.

I'm not intimately familiar with `i87`  code right now having adopted this out-of-tree driver to support a motherboard I acquired, but I've been hacking on the kernel since 2005 so I know my way around enough to be able to code something that'll pass muster in mainline. I'm also familiar with working with chip level interfacing and interpreting and reverse-engineering opaque interfaces.

frankcrawford, 2023-06-29:
TJ,I'm away for a few days, but will get back to you when I get back.ThanksFrankSent from my Galaxy

frankcrawford, 2024-01-03:
@iam-TJ did you ever do anything more about this idea?

iam-TJ, 2024-01-10:
> @iam-TJ did you ever do anything more about this idea?

No. It needs someone very familiar with it87 devices to do the initial design sketch with real example use-cases that can be used to validate and improve the design. With a skeleton framework in place then others would be able to contribute code both in-kernel and for userspace tools.

amfern, 2025-04-27:
@frankcrawford  I would like to help you mainlining the changes in this repo, if that's alright?

Could you please share any development notes, how you test and debug this driver?

frankcrawford, 2025-04-27:
@amfern such help would be appreciated, but the biggest question is, have you submitted patches to the mainline kernel previously.

The issue isn't really development or testing (as that is what most of the users of this repo are doing for us), but rather the bureaucratic process needed to submit patches.  They need to be split into the smallest possible individual patches, so you may end up with a big patchset with multiple patches, and should only be for an specific change.

The other issue is that we still need to get some basic structural changes into the mainline kernel (e.g. handling different numbers of fans in similar chips, etc), and as I was never involved in the writing of them, I need to analyse them prior to submission to make sure I can actually justify them.

Now saying that, it isn't a really hard process, just going to be long and need someone with some time to work on.

If you have time for it, I'm happy to work with you (or anyone) on it so we can get it done.

frankcrawford, 2025-04-27:
I should add this is the official process, [https://www.kernel.org/doc/html/v4.17/process/submitting-patches.html](https://www.kernel.org/doc/html/v4.17/process/submitting-patches.html) although there are number of other documents available online.

Also, the subsytem group is `hwmon` that these patches need to go to.

amfern, 2025-04-27:
Unfortunately, I haven’t submitted a patch to the kernel yet, but I do have the patience and experience to break down large changes into smaller, more manageable patches and gradually get them merged upstream. Since you’ve successfully mainlined some of the changes here, I was wondering if I could ask you to review my patchset before I submit it to the kernel mailing list.

From reviewing the kernel mailing list, this repository, and the kernel source code, it looks like the original author of this patch is @groeck, who also happens to be the maintainer and approver of the hwmon subsystem.
So i am just wondering if you know why he hasn't mainlined the patches?

frankcrawford, 2025-04-27:
@amfern yes, I'm happy to help you through the process, as I've learnt it over time.  I actually know which ones I was aiming for next, but have never got around to it.  Essentially I have been going through the FEATURES, although the next one or two involve the more complex changes.

As for @groeck much of the older work you can find from his original patch list for the kernel upgrade, which was never accepted (I suspect that is when he gave up too), and while he understands what we are doing he is now fully concentrating on keeping the full `hwmon` subsystem clean and not worried about the `it87` module itself.  I also think that the failure to get any support from Gigabyte or ITE also was an issue which caused him to finally give up on this module.

amfern, 2025-04-27:
@frankcrawford i want to learn what arguments @groeck were facing, can you point me to the mailing archive?

mirh, 2025-04-27:
I can't really find the one thread I had in mind (something about other maintainers really not feeling like including such tentative delicate code, unless it was really really tested good by many people) but I found:
https://github.com/lm-sensors/lm-sensors/issues/167#issuecomment-487040285
https://lore.kernel.org/all/e4b5b93e-65ee-dd23-93a6-1737ede87ef4@roeck-us.net/

Also arguably today EC drivers (with or without official datasheets) are all the rage.

amfern, 2025-04-29:
Thank you.

What are the next two features were you thinking to submit?

> e.g. handling different numbers of fans in similar chips, etc

What do you mean handling different part of the chip? Are you talking about this part of the code?
What https://github.com/frankcrawford/it87/blob/4bff981a91bf9209b52e30ee24ca39df163a8bcd/it87.c#L4049

Also what is the "EC" that they mention in the original kernel thread?

mirh, 2025-04-29:
https://en.wikipedia.org/wiki/Embedded_controller
AFAIK it used to be once upon a time that WMI was the preferred way to tinker with hardware voodoo (both because that's probably what windows drivers tended to use, but also because IIRC it avoided you mutexes/synchronization worries or something like that).
But for whatever reason now, not only not all methods/features/operations are exposed through it but exactly the fact that such calls are blocking of everything else is a big no-no (see also the funny saga of asus-wmi-ec-sensors being replaced by asus-ec-sensors just a few months after having been merged).

> What are the next two features were you thinking to submit?

A quick diff shows MMIO access, and multi-bank support.. Which is probably also the answer to your other question.

frankcrawford, 2025-04-30:
P.S. There are a bunch of minor updates for new chips, but the new temperature map and bank selection are the main ones needed for them.