frankcrawford/it87 #8: help with adding support for other ITE chips

ntzb, 2023-04-07:
hi, I have a laptop with an ITE sensors that is probably not from the it87 family.
`sensors-detect` output has:
```
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x5571
```

from this I infer I have the it5571 chip, and no driver exists for it whatsoever.
my question is, can the same principles from the it87 family be applied to this chip?
ITE will not provide the datasheet for this, but can I still try to manually manipulate it to get some results?
e.g. the readme has a passage on unsupported chips, which uses `sudo isadump`, etc (tried to run it, but the output is mostly `00`'s)

would appreciate any guidance.
my fan is blowing too much most of the time and it is driving me crazy..

frankcrawford, 2023-04-16:
@ntzb I'm willing to help, but not sure which way to go here.  I would say that a chip ID of 0x5571 is no likely to be a hardware monitoring one, as they normally have IDs in 0x86XX or 0x87xx range.  That one looks to be very different.

What motherboard is this running on?  What details can you see in the BIOS about the hardware, such as fans, power, temp, etc?

Also, can you also run Windows on this host, as a great source of additional information is the HWiFO package.

Let me know what you see and I'll look at what we can do.

Regards
Frank

ntzb, 2023-04-16:
@frankcrawford thanks.
In various places selling boards with this chip (ITE IT5571) they call it an "Embedded Controller" but I read that it's an interchangeable term with "Super IO".
from my understanding it is related in some why to temp monitoring and fan control.

my laptop is ASUS made, model `TP3402ZA`.
the BIOS has a "System Information" window that includes components names, and a "Fans" entry that include the temperature and fan reading. there's no fan _control_ in the BIOS.
booting Windows, there's fan profile control in the MyASUS app (silent, normal, performance), but HWiNFO and speedfan do not report fan information, only temperature readings.
in HWiNFO I have:
```
Super-IO/LPC Chip:                      ITE IT5571

...

Voltage Probe -------------------------------------------------------------

  Description:                            Voltage Probe #1
  Location:                               Motherboard
  Status:                                 OK
  Maximum Value:                          Unknown
  Minimum Value:                          Unknown
  Resolution:                             Unknown
  Tolerance:                              Unknown
  Accuracy:                               Unknown

Voltage Probe -------------------------------------------------------------

  Description:                            Voltage Probe #2
  Location:                               Motherboard
  Status:                                 OK
  Maximum Value:                          Unknown
  Minimum Value:                          Unknown
  Resolution:                             Unknown
  Tolerance:                              Unknown
  Accuracy:                               Unknown

Cooling Device ------------------------------------------------------------

  Type:                                   Power Supply Fan
  Description:                            Cooling device #1
  Status:                                 OK

Cooling Device ------------------------------------------------------------

  Type:                                   Power Supply Fan
  Description:                            Cooling device #2
  Status:                                 OK

Cooling Device ------------------------------------------------------------

  Type:                                   Unknown
  Description:                            Cooling device #3
  Status:                                 OK

Temperature Probe ---------------------------------------------------------

  Description:                            Temperature Probe #1
  Location:                               Motherboard
  Status:                                 OK
  Maximum Value:                          Unknown
  Minimum Value:                          OK
  Nominal Value:                          OK
  Resolution:                             OK
  Tolerance:                              OK
  Accuracy:                               Unknown

Temperature Probe ---------------------------------------------------------

  Description:                            Temperature Probe #2
  Location:                               Unknown
  Status:                                 OK
[... 61 lines cut ...]
  Tolerance:                              Unknown
  Accuracy:                               Unknown

Electrical Current Probe --------------------------------------------------

  Description:                            Electrical Probe #4
  Location:                               Unknown
  Status:                                 OK
  Maximum Value:                          Unknown
  Minimum Value:                          Unknown
  Resolution:                             Unknown
  Tolerance:                              Unknown
  Accuracy:                               Unknown

Electrical Current Probe --------------------------------------------------

  Description:                            Electrical Probe #5
  Location:                               Unknown
  Status:                                 OK
  Maximum Value:                          Unknown
  Minimum Value:                          Unknown
  Resolution:                             Unknown
  Tolerance:                              Unknown
  Accuracy:                               Unknown
```

frankcrawford, 2023-04-16:
@ntzb

I don't think it is possible to support it with this driver, and it is likely that it doesn't really supply the values you are after anyway.
The SuperIO class of chips basically provide a lot of functions in the one chip, including serial ports, mouse ports, etc, etc, one of which is often HW monitoring.  This driver only supports the HW monitoring part.
However, looking at similar chips that ITE produce (IT5570/IT5576, etc) they do not have the right HW monitoring setup.
You may want to see if any of the maintainers of ASUS specific modules have any idea of what may be possible.

Sorry,
Frank

ntzb, 2023-04-16:
@frankcrawford your input is highly appreciated.
can you please explain what you mean by "they do not have the right HW monitoring setup"?
and also could you point me to the said ASUS modules?

frankcrawford, 2023-04-16:
@ntzb Looking at the spec sheet for the chips I mentioned (e.g. IT5570) vs on of the modules support by the it87 module it does not mention registers accessing the various internal sensors.  Even in the it87 range of chips I know there are some that don't have the hardware sub-chipset.

As for contacts, the MAINTAINERS file lists:

ASUS NOTEBOOKS AND EEEPC ACPI/WMI EXTRAS DRIVERS
M:      Corentin Chary <corentin.chary@gmail.com>
L:      acpi4asus-user@lists.sourceforge.net
L:      platform-driver-x86@vger.kernel.org
S:      Maintained
W:      http://acpi4asus.sf.net
F:      drivers/platform/x86/asus*.c
F:      drivers/platform/x86/eeepc*.c

ASUS TF103C DOCK DRIVER
M:      Hans de Goede <hdegoede@redhat.com>
L:      platform-driver-x86@vger.kernel.org
S:      Maintained
T:      git git://git.kernel.org/pub/scm/linux/kernel/git/pdx86/platform-drivers-x86.git
F:      drivers/platform/x86/asus-tf103c-dock.c

ASUS WMI HARDWARE MONITOR DRIVER
M:      Ed Brindley <kernel@maidavale.org>
M:      Denis Pauk <pauk.denis@gmail.com>
L:      linux-hwmon@vger.kernel.org
S:      Maintained
F:      drivers/hwmon/asus_wmi_sensors.c

ASUS EC HARDWARE MONITOR DRIVER
M:      Eugene Shalygin <eugene.shalygin@gmail.com>
L:      linux-hwmon@vger.kernel.org
S:      Maintained
F:      drivers/hwmon/asus-ec-sensors.c

ASUS WIRELESS RADIO CONTROL DRIVER
M:      João Paulo Rechi Vita <jprvita@gmail.com>
L:      platform-driver-x86@vger.kernel.org
S:      Maintained
F:      drivers/platform/x86/asus-wireless.c

Frank

ntzb, 2023-04-16:
@frankcrawford thanks for the info and guidance

closed 2023-04-16 by ntzb