frankcrawford/it87 #15: Help to get the sensors working for a Gigabyte X670E Aorus Master (AM 5 Board with Ryzen 9)

f45tb00t, 2023-11-20:
Hello Frank,

over the last weeks since I have a new PC built I did quite a lot of research how to get the temperature sensors to work. Just with the stock lm-sensors, nothing is really working. 

I also tried to get to know how I could build a driver, but actually I am a little lost what I need to do.

Maybe you can help, as I would like to provide also to others a working solution. As we are kinda blind when it's about CPU temperatures with this new boards.

Some Questions:

1. How can I determine which precise chips are shipped with the board?
With sensors-detect I found this:
```
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689
Probing for Super-I/O at 0x4e/0x4f
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8883
    (logical device 4 has address 0x290, could be sensors)
```

2. Where can I find the the correct drivers for these chips?

3. How do I build the drivers and make them available as you did?

Thanks a lot for the good work! :)

regards

FB

graysky2, 2023-12-30:
You need to build the it87 module using the code in this repo.  See: https://aur.archlinux.org/packages/it87-dkms-git

frankcrawford, 2024-01-03:
@f45tb00t sorry about my slow response, I've been tied up with other things.

Firstly, `sensor-detect` only correctly handles sensors in the standard kernel module, not the additional ones in this module.

As for how to build it.  For a simple test you can download this repo and just run "make" and then "sudo make install", which will overwrite the existing kernel module.  After that, you may need to reboot for the new module to take effect.

However, for ongoing installation, as you upgrade your kernel, you probably want to use the `dkms` package, and hence you can follow the instructions above.  You do need the `dkms` package installed, and that depends on your specific OS.
Alternatively, if you have `dkms` installed, you can also just run the script `dkms-install.sh` found in this repo, but that may not always work.

Anyway, see how this goes and let me know if it works for you

programminghoch10, 2024-02-07:
Hey @frankcrawford 
i have the exact same board and have been using your kernel module for a while now.
I'd be up to provide required data and testing if you need something.
The mainboard has 8 fans, but the "compatibilty" mode your module uses shows only 5 of them.

This is the current `sensors` output on linux kernel `6.5.0-5-amd64` (debian):
```
it8689-isa-0a40
Adapter: ISA adapter
in0:         756.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in5:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in6:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.29 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.24 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +59.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +57.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +49.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +48.0°C  (low  =  +0.0°C, high = -125.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

frankcrawford, 2024-02-09:
@programminghoch10 okay, I can see the issue, I think, the Gigabyte X670E Aorus Master looks to have 2 chips, with the it8689 only providing 5 fans, and the other would provide the other details.

However, we need to see what the other chip is, so are you able to provide additional details on what chips are in the system?

frankcrawford, 2024-02-09:
@programminghoch10 looking at some other online documentation that board also has an IT8792E (hw ID 0x8733) chip.  If you run sensor-detect does it show anything else  Secondly does anything else appear in the boot log for the it87 module?

programminghoch10, 2024-02-09:
@frankcrawford 
`sensors-detect`:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8883
    (logical device 4 has address 0x290, could be sensors)
```
Yes it seems there are two ITE chips connected.

I don't exactly know what you mean with "boot log", I assume this:
```
$ dmesg | grep it87
[    9.294881] it87: loading out-of-tree module taints kernel.
[    9.294885] it87: module verification failed: signature and/or required key missing - tainting kernel
[    9.295680] it87: it87 driver version v1.0-137-g1663f97.20240207
[    9.295839] it87: Found IT8689E chip at 0xa40, revision 1
[    9.295966] it87: Beeping is supported
```

frankcrawford, 2024-02-10:
@programminghoch10 how confident are you with git, and with making and installing modules into the kernel?

I am thinking of trying a few things in a separate git branch but would need you to compile it and test it, as I don't have that board.

programminghoch10, 2024-02-10:
@frankcrawford absolutely no problem. Bring it on! 👍

frankcrawford, 2024-02-10:
@programminghoch10 when you have a chance can you try the dev branch it87922e in the github repo.

You will probably need to compile it (with "make"), unload the current one ("modprobe -r it87") and load up the local one ("modprobe hwmod_vid ; insmod it87.ko").

If I have it correct, it should detect the second chip as an IT87922E (TBC).  Then secondly, see if the values look to be correct.

programminghoch10, 2024-02-10:
@frankcrawford 
`sensors`:
```
it8689-isa-0a40
Adapter: ISA adapter
in0:         744.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in5:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in6:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.29 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.24 V  
fan1:         167 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +61.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +70.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +51.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +51.0°C  (low  =  +0.0°C, high = -125.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```
still shows only one ITE sensor device

`dmesg | grep it87`:
```
[    7.407978] it87: loading out-of-tree module taints kernel.
[    7.407982] it87: module verification failed: signature and/or required key missing - tainting kernel
[    7.408683] it87: it87 driver version v1.0-137-g1663f97.20240207
[    7.408850] it87: Found IT8689E chip at 0xa40, revision 1
[    7.408977] it87: Beeping is supported
[ 9285.619736] it87: it87 driver version v1.0-142-g58fda1f.20240210
[ 9285.619957] it87: Found IT8689E chip at 0xa40, revision 1
[ 9285.620097] it87: Beeping is supported
[ 9285.624185] it87: Device (chip IT87922E (TBC) ioreg 0x4e) not activated, skipping
```

btw thanks for reminding me that kernel modules can be loaded dynamically, I do know me some linux but I totally forgot about that and thought I would have to reboot after `dkms` installation :eyes:

frankcrawford, 2024-02-11:
@programminghoch10 well the output from dmesg shows it did detect the other chip, but didn't activate it.

It may be an issue that the second chip may need to be configured first, which is a known problem.  To correct it, you may need to reboot your system, and ensure that the new module is the one loaded, not the older one.

programminghoch10, 2024-02-11:
@frankcrawford 
I've made a `dkms` install and rebooted.
The `sensors` is the same, still only one device.
 `dmesg` seems to be the same too:
```
[    5.848180] it87: loading out-of-tree module taints kernel.
[    5.848184] it87: module verification failed: signature and/or required key missing - tainting kernel
[    5.848698] it87: it87 driver version v1.0-142-g58fda1f.20240211
[    5.848902] it87: Found IT8689E chip at 0xa40, revision 1
[    5.849029] it87: Beeping is supported
[    5.853051] it87: Device (chip IT87922E (TBC) ioreg 0x4e) not activated, skipping
```

frankcrawford, 2024-02-11:
@programminghoch10 okay, I'll have to see what I can find out about why it isn't activating.

Thanks.

frankcrawford, 2024-02-12:
@programminghoch10 I've just pushed a new version, although all it does is add some more messaging, so I can see what is going on.

Also, can you send the output from /sys/class/dmi/id/board_name and /sys/class/dmi/id/board_vendor

Thanks.

programminghoch10, 2024-02-13:
@frankcrawford i pulled, `make`d and inserted directly (no `dkms`)

```bash
[14678.472253] it87: it87 driver version v1.0-143-gc51843b.20240213
[14678.472338] it87: Entering config for second chip
[14678.472521] it87: Found IT8689E chip at 0xa40, revision 1
[14678.472656] it87: Beeping is supported
[14678.477131] it87: Device (chip IT87922E (TBC) ioreg 0x4e) not activated, skipping
```

```bash
$ cat /sys/class/dmi/id/board_{name,vendor}
X670E AORUS MASTER
Gigabyte Technology Co., Ltd.
```

frankcrawford, 2024-02-13:
Hmm, the values are coming back right, but not activating properly, so I need to look into further.

frankcrawford, 2024-02-13:
@programminghoch10 I've done a hack to get past the activation issue.  See how it works, but also be aware that this is slighlty dodgy, so be prepared to roll it back.

programminghoch10, 2024-02-13:
@frankcrawford 
```
[ 1895.002060] it87: it87 driver version v1.0-144-gc3fd522.20240213
[ 1895.002092] it87: Entering config for second chip
[ 1895.002269] it87: Found IT8689E chip at 0xa40, revision 1
[ 1895.002397] it87: Beeping is supported
[ 1895.006482] it87: Found IT87922E (TBC) chip at 0x290, revision 1
[ 1895.006568] it87: Beeping is supported
```

`sensors` output is still the same, only one device

f45tb00t, 2024-02-14:
Hi guys,

sorry for my late reply... Life... you know...

It's awesome that you work on this. I'll test as well. 
Meanwhile I tested also other alternatives. What works for me is another project. CoreFreq. 

@programminghoch10 Maybe some things to share with you about the board I found out
1. Be aware that when you upgrade to 4X RAM, the EXPO profiles won't work anymore. I've contacted the GigaByte Support and they told me that when you use full rambanks, that the profiles do not work anymore, as the specification of the CPU does not allow that. I was not aware about that and my DDR5 6k chips cannot be used at full speed. I've now 128GB RAM with a timing of 3600 instead 6000.
2. I've unfortunately issues with the 5.1 sound over HDMI (iGPU). Tried also bleeding edge Ubuntu for instance, but only 2.0 Sound works.

Beside that I am really happy with the setup. With the Ryzen 7600 (non x  version because of 24/7 usage and the power consumption) the pc is a monster :D

frankcrawford, 2024-02-14:
@programminghoch10 Another attempt, based on some comments I've recently been given.

frankcrawford, 2024-02-14:
@f45tb00t that is an interesting item about 4X RAM causing a problem with EXPO profiles.  I think I've seen that previously on other boards.

f45tb00t, 2024-02-14:
> @f45tb00t that is an interesting item about 4X RAM causing a problem with EXPO profiles. I think I've seen that previously on other boards.

It seems this is a common problem. Unfortunately. 

```
Max Memory Speed
2x1R    DDR5-5200
2x2R    DDR5-5200
4x1R    DDR5-3600
4x2R    DDR5-3600
```
Ref: https://www.amd.com/de/product/12746

To be honest I was not aware about that. I think that would have changed my "buying-mind" a little. Because why should I buy then faster RAM for much more money, if they just do not run as expected. I did not try to get the maximum out by manually tweaking, as I need a reliable and stable production system.  Furthermore I am not in that tweaking anymore. I don't play games with this pc, so therefore I don't care about 0,00001% more speed.  But surely the EXPO was an interesting thing. My last PC was like 10+++ years old, still running, but got very slow over the last years due to the CPU vulns and the applied fixes. So I decided to create a new one without any compromises this time. 

Once I got the temperatures of the Cores and fix the 5.1 sound issues, I'll be totally happy with this setup. Hope it lasts also for 10+++ years. :)

frankcrawford, 2024-02-15:
@programminghoch10 so, talking with someone who has some knowledge of the chips involved, it turns out we have this all wrong.  What he said is

_0x8883 is IT8883FN which is just a plain eSPI to LPC Bridge, more-less transparent on the bus. Behind it usually lies the above chip.
0x8883 should normally not appear on the LPC bus but it can happen if the bug I described below occurs._

The bug he is describing is that the chip locks up if not accessed correctly, which I thought we did, but may not be done correctly.

Anyway, firstly try the version I pushed recently, but expect more changes, to try and get it working properly.

programminghoch10, 2024-02-15:
@frankcrawford 
```
[  147.963350] it87: it87 driver version v1.0-145-gc635724.20240215
[  147.963387] it87: Entering config for second chip
[  147.963530] it87: Found IT8689E chip at 0xa40, revision 1
[  147.963657] it87: Beeping is supported
[  147.967617] it87: Device (chip IT87922E (TBC) ioreg 0x4e) not activated, skipping
```

@f45tb00t thanks for sharing your experience with the mainboard
i also got quite the time with it and have a lot to tell,
however im a bit short on time currently, will properly tell you all i found out at a later date
just a quick question though, do you have mainboard of `rev 1.0` or `rev 1.1`?

frankcrawford, 2024-02-15:
@programminghoch10 when you do have some time, let me know and I'll get you to try something else more to see the results.

programminghoch10, 2024-02-16:
@f45tb00t there is and i have one after quite some issues with rev 1.0
I'll tell you more at a later date.

@frankcrawford 
You can keep working on it, i have the 5 minutes to compile and run it on the evening.
I just meant that i don't currently have the time to write out the extensive story for @f45tb00t .

frankcrawford, 2024-02-16:
@programminghoch10 I am going to keep working on it, but will do a bit of restructuring, which will affect not just you but other chips as well.

However, what I would like you to try, when you have a chance is to:
a) blacklist the it87 module (depending on system is something like adding "blacklist it87" into /etc/modprobe.d/blacklist)
b) reboot
c) run "sudo sensors-detect" to see what is then listed.
d) then un-blacklist it87

The reason for this is I want to test if it is the it87 module that is crashing the second chip or something else.  If it is the it87 module, you should no longer see the chip ID 0x8887, but a different chip ID (I think).

frankcrawford, 2024-02-16:
Oh, rats, I just checked the sensor-detect module and it may be running the bad code as well.  Anyway, we will see.

frankcrawford, 2024-02-16:
Just changing my request please do the following:

a) blacklist the it87 module (depending on system is something like adding "blacklist it87" into /etc/modprobe.d/blacklist)
b) reboot
c) run "sudo isadump 0x4e 0x4f 7" (hopefully there won't be lots of "ff" entries)
d) run "sudo sensors-detect" to see what is then listed
e) then un-blacklist it87

Thanks.

frankcrawford, 2024-02-16:
@programminghoch10 After you try the above tests see how the new code I've just pushed works.

programminghoch10, 2024-02-17:
@frankcrawford 
`isadump`:
```
$ sudo isadump 0x4e 0x4f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 
20: 87 33 03 00 00 00 ff 01 3e 18 02 00 8b 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 0a 50 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 20 38 00 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 01 00 01 3e 08 00 00 00 00 00 00 14 08 00 1e 60 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 20 20 20 c4 3e 3e 0c 
f0: 00 00 00 00 00 00 2c 00 00 00 00 00 ff 00 00 00
```

`sensors-detect`:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8883
    (logical device 4 has address 0x290, could be sensors)
```

then i inserted the module (without rebooting), still the same `sensors` output.
here it is again for reference:
```
it8689-isa-0a40
Adapter: ISA adapter
in0:           1.45 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in5:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in6:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.29 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.24 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +28.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +57.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +33.0°C  (low  =  +0.0°C, high = -125.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

frankcrawford, 2024-02-17:
@programminghoch10 that result is both good and bad.  Firstly, the isadump at offset 20 shows the chip ID and is is 8733, which is a valid IT87 chip, not 8883, which is causing us trouble, so it does look like it is getting broken by the initialisation somehow.

So, try the following:
a) blacklist the it87 again
b) reboot
c) run isadump
d) run sensor-detect
e) run isadump (offset 0x20 should be now different)
f) reboot
g) insmod (or modprobe) the it87 module (without doing anything else at all before hand).

Also, make sure to grab the latest git version, I did late last night.

Thanks
Frank

programminghoch10, 2024-02-17:
@f45tb00t 
got a bit of time now, here's my story:

I got this mainboard when it was fairly new, around autumn 2022.
It never worked flawlessly, had a bunch of issues, but since the issues _changed_ with BIOS updates I thought it might sort itself out.
My 4x 32GB RAM sticks also wouldn't work on `5200` EXPO, only on `3600` which I back then thought was odd, because they were labelled as at least `4800`.

However mid 2023 a weird issue appeared which I couldn't figure out:
The mainboards BIOS was running completely in what I can only describe as "Slow-Motion". The BIOS took about 3 minutes to initialize, and the booting sequence images (like the ati logo and stuff) were visibly taking longer than usual. When booting in this state the whole OS would also be rendered about useless, however resetting sometimes up to 3 times in a row made it possible to boot up normally. 
After BIOS updates didn't solve this issue at all and Gigabyte eSupport also wasn't able to help I sent in the mainboard for warranty/repair. Was quite the ordeal to get it out of the case 🙄 
Of course they didn't repair it, instead they sent me a `rev 1.1`.
You can see [on their specifications site](https://www.gigabyte.com/de/Motherboard/X670E-AORUS-MASTER-rev-1x/sp#sp) that these revisions officially only differ in their WiFi chipset, now featuring a `Mediatek MT7922A22M` instead of `Intel AX210` (the mediatek drivers on linux are a bit more flakey than the intel ones i feel, at least for bluetooth, I don't use WiFi)
However they must have changed something else as my "slow motion" issue didn't appear again _(yet)_.

So me now being on the new revision I retested all the stuffs. Here are my results mixed with what I found out on the previous board too. 
* 4 RAM will produce `3600` as before, however the system doesn't crash anymore when I enable the EXPO profile, so I am currently actually running 4 slot EXPO with `5200`.
* Bluetooth and WiFi use the same antenna, I thought I wouldn't need it since I don't use WiFi, but the bluetooth range without antenna is only about 30cm, so I would advise you to install the antenna even if you don't use it regularly.
* You cannot do legacy boot with iGPU. It just doesn't work. It should, but it doesn't. Maybe a BIOS update will fix this.

Regarding fan control:
At first I was a bit sad I couldn't control the fans with `fancontrol` as I did on my previous computer (but maybe we are able to change that at some point :eyes:) however I gotta say the fan settings in the BIOS are quite good and I was able to set it up as I wanted completely in the BIOS. If you want fans to stop `voltage`-based control mode is the one to go for, as even PWM fans will stop then. 
I have the big noctua cooler thing (`NH-D15`) and I currently have my fans set up to start spinning at about 70°C CPU temp, 75°C for the case fans.
This way while idling at the PC the CPU fans mostly spin at minimum speed or not at all, while everything ramps up with increased load.
If I limit the processor speed to 3GHz, the fans never ramp up, even under full load. 3GHz on 32 threads is still plenty enough for everything.
If you want I can provide you my processor speed control bash function.

I also had some issues installing to NVMe but since I didn't feel like re-installing the whole OS anyways, I've been using my old SATA SSD install for over a year, so I can't say much about that. But I'll definitely switch to NVMe at some point, currently the NVMe SSD is just sitting around and collecting "Power-On Hours" lol.

I also did a quick reboot while writing to document my BIOS settings for you (the ones I changed from default setting):
* `Tweaker > Advanced CPU settings > SVM mode`: `Enabled`
   This enables the use of hardware accelerated virtualization / `KVM`
* enabled `EXPO` profile
* `Settings > IO Ports > Gigabyte Utilites downloader`: `Disabled`
* `Settings > IO Ports > Integrated graphics`: `Enabled`
   `nvtop` crashes with the iGPU enabled, also I don't need it anyways
* `Boot`
  * `Fast boot` off
  * `CSM support` off (might be default already)
  * `Secure boot` off
  * `Full screen logo show` off
  * `Bootup Numlock state` on

I also completely turn of my `TPM`. I don't have an encrypted setup yet, but the `fTPM` this mainboard has is not well received by the linux developers, thinking about buying an SPI TPM before doing an encrypted install.

I currently have the BIOS version `F12`, the one my `rev 1.1` mainboard came with.

Hope you found it interesting :)

programminghoch10, 2024-02-17:
@frankcrawford 
module blacklisted, rebooted,
`isadump 0x4e 0x4f 7`:
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 
20: 87 33 03 00 00 00 ff 01 3e 18 02 00 8b 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 0a 50 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 20 38 00 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 01 00 01 3e 08 00 00 00 00 00 00 14 08 00 1e 60 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 20 20 20 c4 3e 3e 0c 
f0: 00 00 00 00 00 00 2c 00 00 00 00 00 ff 00 00 00 
```
`sensors-detect`:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8883
    (logical device 4 has address 0x290, could be sensors)
```
I only did the sensors detect until this part showed up, then i left with ctrl-c.

`isadump 0x4e 0x4f 7` directly after `sensors-detect`:
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```
Definitelfy something fishy there.

I kept the module blacklisted, then inserted the module after booting:
```
[   63.211781] it87: it87 driver version v1.0-146-g7f93a2a.20240217
[   63.211961] it87: Found IT8689E chip at 0xa40, revision 1
[   63.212089] it87: Beeping is supported
[   63.216100] it87: Found IT8792E/IT8795E chip at 0xa60, revision 3
[   63.216185] it87: Beeping is supported
```

btw I always pull and build the `it87922e` branch before testing, dont worry

`sensors`
```
it8689-isa-0a40
Adapter: ISA adapter
in0:           1.43 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in5:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in6:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.22 V  
fan1:         166 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +59.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +69.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +51.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +53.0°C  (low  =  +0.0°C, high = -125.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

it8792-isa-0a60
Adapter: ISA adapter
in0:           1.78 V  (min =  +0.00 V, max =  +2.78 V)
in1:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
in2:           1.82 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:         3.36 V  (min =  +0.00 V, max =  +5.56 V)
in4:           1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.04 V  (min =  +0.00 V, max =  +2.78 V)
in6:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:          3.33 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:          3.27 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +44.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +51.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

Wow two chips are present! Nice!

programminghoch10, 2024-02-17:
also quick and dirty test, I did a full load stress test on the CPU and I can see 4 fans spinning in the case.
`sensors` _(shortended)_:
```
it8689-isa-0a40
fan1:        1161 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:        1209 RPM  (min =    0 RPM)

it8792-isa-0a60
fan1:           0 RPM  (min =    0 RPM)
fan2:         894 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
```
so were still losing a fan sensor somewhere, but I  guess this will need some more extensive testing
If I have more time I can go ahead and replug fans and test how the reported fans map to the board connectors.

frankcrawford, 2024-02-17:
@programminghoch10 that is great, although missing a fan is a bit annoying.  However, it is easier to find one, if it is missing than a whole sensor.

It is likely that it is a different offset for that fans than I currently have, and if you are sure you have values, it should be easy to find it.  I'll send some details a bit later.

I am also interested in the temperature and voltages, to make sure we have them correct too, if you can send them over some time.

It may be worth looking at the values in the sensor conf files in "Sensor Configs", as it is probably close to the fan values you would see on your board.

frankcrawford, 2024-02-18:
@programminghoch10 when you have a chance, we will see if we can find your missing fan.  To do this, send through two more dumps:

isadump 0xa45 0xa46
isadump 0xa65 0xa66

Note, this doesn't need a reboot or anything.  It should just be looking up the same registers as the driver does.

Thank.

programminghoch10, 2024-02-18:
@frankcrawford 
```
$ isadump 0xa45 0xa46 && isadump 0xa65 0xa66
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0xa45 and data register 0xa46.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 10 00 80 df ff 00 20 ff 98 48 40 30 ff ff ff 
10: ff ff ff 77 c7 d0 c0 c0 ff ff ff ff ff ff ff ff 
20: 3e a7 a7 a9 5f 5f a6 89 87 29 3b 41 32 32 c9 80 
30: ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 
40: 7f 7f 7f 7f 7f 7f 5f 40 ad 6a d4 00 00 00 00 00 
50: ff 18 7f 7f 7f 40 00 00 90 00 28 12 65 00 00 00 
60: 00 37 64 00 00 03 00 c0 00 00 00 00 00 01 00 ff 
70: 00 00 00 00 00 01 00 2e 00 00 00 00 00 01 00 c0 
80: ff ff ff ff ff ff ff ff 00 00 00 00 00 39 00 4e 
90: 00 00 00 00 ff 10 10 10 42 99 00 00 01 4d f2 41 
a0: 00 3c 64 00 00 03 00 d0 7f 7f 96 80 00 00 0f 98 
b0: 4e 3a 82 76 7f 7f 83 00 7f 7f 80 29 80 80 80 00 
c0: 00 c0 00 10 00 01 10 10 10 10 90 90 90 90 90 90 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00 00 
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0xa65 and data register 0xa66.
Continue? [Y/n] 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 10 42 00 ff ff 00 00 00 80 48 00 30 ff ff ff 
10: ff ff ff 77 c0 00 00 02 ff ff ff ff ff ff ff ff 
20: a3 ff a6 9a a3 5f ff 99 96 2b c9 33 80 6f 76 76 
30: ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 
40: 7f 7f 7f 7f 7f 7f 5f 40 ad 6a d4 00 ff ff ff ff 
50: ff 38 7f 7f 7f 40 00 00 90 00 09 12 65 00 00 00 
60: 00 14 41 ff 1c 03 00 ff 00 14 41 00 1c 03 00 ff 
70: 00 14 41 ff 1c 03 00 ff ff ff ff 80 ff ff ff ff 
80: ff ff ff ff ff ff ff ff 00 00 00 00 00 ac 00 71 
90: ff 00 00 00 ff 00 00 00 10 97 00 00 40 dd c1 00 
a0: 7f 76 91 dd 34 ff af e8 ee 54 fa db 3f 2f 00 00 
b0: 54 58 58 5a 58 5a 01 01 ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

i don't know what you mean by 
> I am also interested in the temperature and voltages, to make sure we have them correct too, if you can send them over some time.

How am I supposed to know whether the shown values are correct?
Only thing i can think of is noting down the values shown in the BIOS, but then I also can only tell you which values look similar, not the exact mapping, as there are multiple values in the same range.
Which values do you want exactly?

> It may be worth looking at the values in the sensor conf files in "Sensor Configs", as it is probably close to the fan values you would see on your board.

And I also don't know which sensor conf files you mean.

frankcrawford, 2024-02-19:
@programminghoch10 thanks for those dumps, I'll look at them a bit later.

As for the comments, for the voltage and temp, it is just to see if they look reasonable, by comparing what you see in BIOS, or on Windows with things like hwinfo.  It is only to see if they look right, not if t hey are 100% accurate, as I don't know how you can do that either.

Note, especially for the voltages, the mappings are important, as they scale the voltages depending on what type they are.  The temperatures don't normally need any scaling.

As for the sensor scalings, in the folder "Sensor configs" as part of the git repo.  If you look in any of the con files, they list the chips on that server and the equivalent names of the sensor items.  While they don't need to be exactly right, Gigabyte are fairly consistent, so on one board for a certain chip they will tend to match another board too.  Note that the names are based on the motherboard, but sensors doesn't care about that, it only cares about the chips listed.

Look for the entries like chip "it8792-*" or chip "it8689-*", but also chip "it8792-isa-0a60" is fine for your board.  Something like GA-Z690-AORUS-PRO.conf is pretty comprehensive, but has some slightly different chips as well.

Feel free to copy one of those files (or generate a whole new one) for your board and load it in /etc/sensors.d, and it will then be used automatically by sensors.

frankcrawford, 2024-02-19:
@programminghoch10 I've tried a small update to the fan measurements, so, please give it a go.

Also, what connectors do you have the fans connected to?  What sort of fans are they?

programminghoch10, 2024-02-20:
@frankcrawford ive tested it, still the same, 4 fans spinning, 3 PWM sensors showing activity.
regarding creating sensor configs, that seems to be a lot of work, ill consider it in 1-2 months when i have more time.

@f45tb00t 
update on my mainboard situation:
i forgot to say the mainboard time is always wrong on boot
also my PC just didnt start at all today, and i had to disable my `5200` EXPO to get it working again 🙄

frankcrawford, 2024-03-02:
@programminghoch10 since this seems to be working correctly and not breaking other chips, I've now rolled this into the mainline.

Going forward I probably won't keep updating the it87922e branch unless something dramatic happens.

dr3st, 2024-06-16:
Hey ho @programminghoch10 @frankcrawford 

thank you very much for working on this issue, I really appreciate your work!
I use this board as well and got some issues, maybe you can tell if you can confirm this / got an idea how to fix.

I compiled the kernel by archlinux AUR (dkms is used), rebooted and some fans are found, but not every.
If someone else getting into this issue, i mapped mainboard connector to chip / fan id (sensors output), so you can decide to use the correct (supported) pins atm :)

Chip | FAN | Board
-- | -- | --
it8689-isa-0a40 | fan5 | cpu_opt
it8689-isa-0a40 | fan1 | cpu_fan
it8689-isa-0a40 | fan2 | sys_fan1
it8689-isa-0a40 | fan3 | sys_fan2
it8689-isa-0a40 | fan4 | sys_fan3
it8792-isa-0a60 | fan3 | sys_fan4
it8792-isa-0a60 | fan1 | sys_fan5_pump
it8792-isa-0a60 | fan2 | sys_fan6_pump
not-found |   | sys_fan7_pump
not-found |   | sys_fan8_pump

Please do not get irritated by sensors output, for getting the mapping table I used a lose fan on free ports ...
These are my sensors I get with connected fans:
```
it8689-isa-0a40
Adapter: ISA adapter
in0:           1.03 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.04 V  (min =  +0.00 V, max =  +3.06 V)
in3:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.04 V  (min =  +0.00 V, max =  +3.06 V)
in5:           1.14 V  (min =  +0.00 V, max =  +3.06 V)
in6:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.29 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.14 V  
fan1:         968 RPM  (min =    0 RPM)
fan2:         799 RPM  (min =    0 RPM)
fan3:        1439 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:         950 RPM  (min =    0 RPM)
temp1:        +39.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +34.0°C  (low  =  +0.0°C, high = -125.0°C)  sensor = thermistor
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  OK

it8792-isa-0a60
Adapter: ISA adapter
in0:           1.78 V  (min =  +0.00 V, max =  +2.81 V)
in1:           2.81 V  (min =  +0.00 V, max =  +2.81 V)  ALARM
in2:           1.83 V  (min =  +0.00 V, max =  +2.81 V)
+3.3V:         3.39 V  (min =  +0.00 V, max =  +5.61 V)
in4:           1.78 V  (min =  +0.00 V, max =  +2.81 V)
in5:           1.04 V  (min =  +0.00 V, max =  +2.81 V)
in6:           2.81 V  (min =  +0.00 V, max =  +2.81 V)  ALARM
3VSB:          3.37 V  (min =  +0.00 V, max =  +5.61 V)
Vbat:          3.26 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:         859 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +46.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

Is there any chance to give you more information to find out which address to read to get other pins working as well?
The mainboard offers:
- 1 x CPU
- 1 x CPU_OPT
- 4 x SYS_FAN (sys_fan1, sys_fan2, sys_fan3, sys_fan4)
- 4 x SYS_FAN_PUMP (sys_fan5_pump, sys_fan6_pump, sys_fan7_pump, sys_fan8_pump)

frankcrawford, 2024-06-17:
@dr3st Thanks for the report, and even more interesting the comment that the maybe they are using USB for the other fans.  However, I am seeing what I can find with the current chipsets.

The item about RGB LED Controller is more related to the ability to flash lights on the fan, etc, not directly to the management of the fan itself.  There is a package that allows you to manage it, [https://github.com/liquidctl/liquidctl]( https://github.com/liquidctl/liquidctl) which seems to work well.

BBQigniter, 2024-12-23:
Hello,

please, I need help. I have a Gigabyte X870E AORUS PRO which seems to use the same or similar iTE chips and I got at least the 8696 main-chip working on Fedora 41 (Kernel 6.12.5) via installing it with dkms using the latest master branch. This one works like a charm and I'm very happy that I can control all the connected fans.

But I cannot get the second chip working, no matter what I try. 

`dmesg` looks like this:

```
[Mon Dec 23 11:13:29 2024] it87: loading out-of-tree module taints kernel.
[Mon Dec 23 11:13:29 2024] it87: module verification failed: signature and/or required key missing - tainting kernel
[Mon Dec 23 11:13:29 2024] it87: it87 driver version v1.0-169-g213db3a.20240922
[Mon Dec 23 11:13:29 2024] it87: Found IT8689E chip at 0xa40, revision 0
[Mon Dec 23 11:13:29 2024] it87: Beeping is supported
```

My sensor-detect output:

```
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8696
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8883
    (logical device 4 has address 0x290, could be sensors)
```

It would be really nice to get the second chip running because now I have two temperature-sensors connected (in the [mainboard's manual noted as EC_TEMP1/EC_TEMP2 ](https://download.gigabyte.com/FileList/Manual/mb_manual_x870e-aorus-pro-ice_1002_e.pdf?v=59cc1f2d26534159d27e2e7d71214a42)), that seem also to be handled by the second chip :worried:

frankcrawford, 2025-01-11:
@BBQigniter so, the other chip is a new one from ITE, but there is another issue, #40 where it has been tracked, and I'll try and add it into this driver soon.

BBQigniter, 2025-01-12:
thank you very much!

frankcrawford, 2025-03-23:
I'm closing this off now as the intems to follow are in another issue, but feel free to reopen it or open a new issue if there is something to follow up.

closed 2025-03-23 by frankcrawford