frankcrawford/it87 #18: Have to reinstall after every reboot

kada49, 2024-03-03:
Hey, I wanted to ask some clarificaton about which installation method to use as preferred. If I read the README correctly there are three ways to install the module:
1. sudo ./dkms-install.sh
2. make clean && make && sudo make install
3. sudo make dkms

I tried all of these out and came to the result that 2. does not work and 1. and 3. work but after a reboot not anymore.

Note: by "work" I mean that the sensors show up i.e. if I type the command `sensors` and fans and sensors are shown.
Eg. :
```
it8613-isa-0a30
Adapter: ISA adapter
in0:           1.44 V  (min =  +2.08 V, max =  +1.98 V)  ALARM
in1:           1.39 V  (min =  +0.89 V, max =  +0.85 V)  ALARM
in2:           2.00 V  (min =  +2.30 V, max =  +1.71 V)  ALARM
in4:           1.72 V  (min =  +0.68 V, max =  +2.74 V)
in5:           2.00 V  (min =  +2.28 V, max =  +0.33 V)  ALARM
3VSB:          3.28 V  (min =  +2.53 V, max =  +3.41 V)
Vbat:          2.57 V  
+3.3V:         3.28 V  
fan2:        2257 RPM  (min =   30 RPM)
fan3:         799 RPM  (min =   15 RPM)
temp1:        +29.0°C  (low  = +71.0°C, high = +60.0°C)  sensor = thermistor
temp2:        +28.0°C  (low  =  -6.0°C, high = -99.0°C)  ALARM  sensor = thermistor
temp3:        +29.0°C  (low  = -21.0°C, high = -34.0°C)  ALARM
intrusion0:  ALARM
```

About making it "work" after rebooting: Am I missing something I need to add somewhere?
Thanks in advance :)

frankcrawford, 2024-03-05:
@kada49 what distribution are you using?  The issue sounds like a problem with either your installation of DKMS or modprobe.

Also, just going through the installation methods you have tired, they are all related.

1. _sudo ./dkms-install.sh_ sets some environment variable and then calls _make dkms_.  The environment variables allow better control if it isn't the current kernel.
2. _sudo make dkms_ sets up the installation in the appropriate directory for DKMS to compile the installation, which essentially runs _make clean && make && sudo make install_ (not quite as a single command).
3. _make clean && make && sudo make install_ does the actual installation.

However, I think the biggest difference is that the DKMS install finishes with a _modprobe it87_ and that may be why you are not getting it installed after a simple _make_.

Also, next time you reboot, just try _modprobe it87_ and see if that fixes your issue.  If that doesn't work, see if there is a directory _/usr/src/it87-v1.0.*_ (the actual value depends on which git commit you have).

Let me know how it goes.

kada49, 2024-03-05:
Thank you very much, `sudo modprobe it87` "installs" the module. Though I would like to run this before ie fancontrol.service is started, where can I set that? (Fedora 39 btw.)

g6avk, 2024-03-05:
@kada49 on this F39 box I created an it87.conf file which contains the text:

 it87 

and placed this in /etc/modules-load.d/ this way the module will load automatically.

HTH

frankcrawford, 2024-03-06:
@kada49 yes, try what @g6avk suggested, as it will force it to load during booting.

It should be found during the auto-probe during boot, but obviously something is not working properly, but that is a bit difficult to diagnose remotely.

frankcrawford, 2024-03-07:
@kada49 that is really a problem with the driver, and that chipset does look to have some strange configuration in the driver (predating my work).

What motherboard do you have, and are you able to get some additional dumps out of your system, if I give you some other commands to try?

kada49, 2024-03-07:
> What motherboard do you have, and are you able to get some additional dumps out of your system, if I give you some other commands to try?

Sure, I could do that!
Motherboard:
`Manufacturer: MEDION, Product Name: B360H4-EM` by `sudo dmidecode -t 2`

kada49, 2024-03-07:
[image]
This is the only part in the bios where semsors are shown. Advanced -> PC Health Status

kada49, 2024-03-07:
I have Windows too

frankcrawford, 2024-03-07:
In that case, download hwinfo64 (or 32) from https://www.hwinfo.com/ and see what values it reports about the it8613.

The portable version is probably all you need, just to see what is seen under Windows and a totally separate implementation.

kada49, 2024-03-07:
Seems like it got corrupted somehow (Windows boot partition), fixing it and then doing that. May take some time

kada49, 2024-03-07:
[image]
There is no defalut alert low or high / min or max temp/voltage. I can setthese here though:
[image]

frankcrawford, 2024-03-07:
Okay, thanks.  I'll give you a few other things to try over the next few days to see what the underlying values are.

frankcrawford, 2024-03-24:
@kada49 sorry about the delay in getting back to you.  Could you run the following two commands:

`sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7`
`sudo isadump 0x295 0x296`

kada49, 2024-03-24:
Hey, because of the easter break I will not be able to do this soon, in about 2 weeks I will be back at my pc. I hope this is not too much of a problem...

frankcrawford, 2024-03-24:
Sure, that is fine.  Thanks.

kada49, 2024-04-07:
Sorry for the long wait, here is the output

`sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7`:
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 13 01 49 00 08 f7 00 0a 00 01 48 41 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 10 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 09 20 38 00 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 c0 03 00 00 
c0: 01 00 00 40 00 00 00 00 01 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 17 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 07 07 07 00 00 40 0c 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 7c 00 00 00
```

`sudo isadump 0x295 0x296`:
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

frankcrawford, 2024-04-27:
@kada49 thanks for that, I'll see what I can see from this, although obviously the second one is getting no entry, so probably needs to be a different address.

kada49, 2024-06-18:
Any update on this?

frankcrawford, 2024-06-20:
@kada49 sorry, not yet, I have been off on other tasks.

frankcrawford, 2025-03-23:
@kada49 unfortunately I probably don't have much to add with this and no real idea what way forward.

I'm going to close this off, but if you have any ideas, feel free to reopen or open a new issue with additional info.

closed 2025-03-23 by frankcrawford