frankcrawford/it87 #20: Something is wrong with th it8625e support.

shamefulCake1, 2024-03-30:
The $SUBJ chip seems to be supported, and there are a lot of mentions of it in the two patches-differences between this fork and the mainline.

However, even though the module loads successfully and find the chip:

```
[Mar30 22:34] it87: it87 driver version v1.0-154-gf40148e.20240306
[  +0.000316] it87: Found IT8625E chip at 0xa20, revision 12
[  +0.000056] it87: Beeping is supported
```, the `sensors` command does query it:

```
>sensors
sensors
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  
temp2:        +29.8°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
```,

no mention of the `it-8625-*` anything chip.

Is there some kind of a trivial switch which should be turned on?

frankcrawford, 2024-04-27:
@shamefulCake1 sorry, I've been busy for the last month or two and haven't been checking issues.

Anyway, I think your issue is that you has is a fairly simple issue, you are probably running into ACPI blocking loading, and you can see this in the kernel messages just following the ones above.

A quick test is to try `modprobe it87 ignore_resource_conflict=1`

If that works, see my comments I've just added to issue #21 for a fix on boot.

A couple of other items from the link you listed, it looks like manjaro has two separate it87 packages, and most modern chips are probably only in the package based on my repo (it87-frankcrawford-dkms-git ?).  However, you do seem to have loaded, so that isn't an issue.

Secondly, unfortunately `sensor-detect` only detects chips that are in the official kernel, not in my repo, and that is not something I am currently chasing, as I'm trying to get the official module updated first (which is a not a simple process).

shamefulCake1, 2024-04-27:
>A quick test is to try modprobe it87 ignore_resource_conflict=1

No, this does not help. Of course I tried it before. Same effect: the module loads correctly, but `sensors` produces no additional output (even with bogus numbers).

frankcrawford, 2024-04-27:
@shamefulCake1 can you run `lsmod | grep it87` and make sure that the it87 module is loaded.

shamefulCake1, 2024-04-27:
Yes, it is loaded, and dmest still says:

``1
[763650.554094] it87: it87 driver version v1.0-156-gdcc0c5f.20240427
[763650.554197] it87: Found IT8625E chip at 0xa20, revision 12
[763650.554228] it87: Beeping is supported
```

frankcrawford, 2024-04-27:
But the check is that the module is still loaded, hence testing with lsmod.

Secondly, can you try `cat /sys/class/hwmon/hwmon*/name` and see what sensors the kernel thinks exist.

frankcrawford, 2024-04-27:
BTW, what are the lines just after it87 in dmesg?

shamefulCake1, 2024-04-27:
```
root@pi:/etc/modprobe.d# lsmod | grep it87
it87                   73728  0
hwmon_vid              16384  1 it87
2024-04-27T18:35:27+08:00|0|.024201|
root@pi:/etc/modprobe.d# cat /sys/class/hwmon/hwmon*/name
acpitz
coretemp
```

frankcrawford, 2024-04-27:
What motherboard do you have, what distribution are you running?

shamefulCake1, 2024-04-27:
Intel NUC J3455, cpu Intel(R) Core(TM) i3-4012Y CPU @ 1.50GHz

Slackware 15.0

it87 from dkms, from this repo

frankcrawford, 2024-04-27:
Hmm, unusual combination.  How much do you know about the hardware itself?  Do you know if Intel have actually implemented the hwmon functions of the IT8625E?

SuperIO chips (of which is one) implement many hardware low-level functions in the one chip, but nothing says that they all have to be used on every board.

Does the board have an onboard fan?

shamefulCake1, 2024-04-27:
>Do you know if Intel have actually implemented the hwmon functions of the IT8625E?

No, I don't.

>Does the board have an onboard fan?

Well, not in stock. This was actually a problem, since this machine used to overheat to the point of hanging the it87 watchdog function. (I.e., it hang, and even the it87 watchdog, which I am sure is present, couldn't reboot it.)

This is actually why I wanted to investigate this driver, because I thought that this would let me monitor the watchdog chip's temperature instead of the cpu temperature.

The BIOS utility has the interface for the fan control, so the motherboard might actually support a fan, it's just that the device didn't have one in stock.

frankcrawford, 2024-04-27:
Can you also run `sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7` and let me see what comes up.

shamefulCake1, 2024-04-27:
```
isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 25 0c 49 00 00 00 00 00 01 01 48 41 06 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 00 80 78 00 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 c0 03 00 00 
c0: 01 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 b0 44 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 1d d1 dc 00 00 40 04 
f0: 00 00 00 00 00 00 0e 00 00 00 00 00 7c 00 00 00 

```

frankcrawford, 2024-04-27:
Give me some time to analyse that to see what it is reporting.

shamefulCake1, 2024-04-27:
Absolutely. If can somehow help, just tell me.

amfern, 2025-04-27:
I have the BIOSTAR 650Q Racing board, and i had to enable the `IT8625E` in the lm_sensors source code like so.

```
From 5eba2633b316aa0a24c1557bee8baf7ec3d91a91 Mon Sep 17 00:00:00 2001
From: Ilya Guterman <amfernusus@gmail.com>
Date: Mon, 3 Feb 2025 01:47:12 +0900
Subject: [PATCH] wip

---
 prog/detect/sensors-detect | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/prog/detect/sensors-detect b/prog/detect/sensors-detect
index 4a73e1f9..8d2c8c09 100755
--- a/prog/detect/sensors-detect
+++ b/prog/detect/sensors-detect
@@ -2540,7 +2540,7 @@ use constant FEAT_SMBUS   => (1 << 7);
        features => FEAT_IN | FEAT_FAN | FEAT_TEMP,
    }, {
        name => "ITE IT8625E Super IO Sensors",
-       driver => "to-be-written",  # it87
+       driver => "it87",   # it87
        devid => 0x8625,
        logdev => 0x04,
        features => FEAT_IN | FEAT_FAN | FEAT_TEMP,
--
2.49.0
```

And now i am able to see the temperatures.
```
$ sensors

it8625-isa-0a20
Adapter: ISA adapter
in0:           1.26 V  (min =  +1.04 V, max =  +2.81 V)
in1:           1.36 V  (min =  +0.00 V, max =  +2.81 V)
in2:           2.04 V  (min =  +0.00 V, max =  +2.81 V)
in3:           2.04 V  (min =  +0.00 V, max =  +2.81 V)
in4:           1.14 V  (min =  +0.00 V, max =  +2.81 V)
in5:           1.79 V  (min =  +0.00 V, max =  +2.81 V)
in6:           1.30 V  (min =  +0.00 V, max =  +2.81 V)
3VSB:          3.34 V  (min =  +0.00 V, max =  +5.61 V)
Vbat:          3.23 V
+3.3V:         3.26 V
fan1:         695 RPM  (min =   11 RPM)
fan2:         420 RPM  (min =   11 RPM)
fan3:         590 RPM  (min =   17 RPM)
fan4:           0 RPM  (min =   -1 RPM)  ALARM
fan5:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +91.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermal diode
temp2:        +59.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +52.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

spd5118-i2c-16-51
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +53.0°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

amdgpu-pci-1000
Adapter: PCI adapter
vddgfx:        1.31 V
vddnb:         1.24 V
edge:         +50.0°C
PPT:          63.07 W

nvme-pci-0400
Adapter: PCI adapter
Composite:    +63.9°C  (low  =  -0.1°C, high = +85.8°C)
                       (crit = +86.8°C)
Sensor 1:     +57.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +59.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +91.0°C

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +46.0°C

spd5118-i2c-16-53
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +52.0°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +63.6°C
Tccd1:        +53.4°C
Tccd2:        +47.5°C

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      314.00 mV
fan1:           0 RPM  (min =    0 RPM, max = 3300 RPM)
edge:         +48.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +52.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +66.0°C  (crit = +108.0°C, hyst = -273.1°C)
                       (emerg = +113.0°C)
PPT:          15.00 W  (cap = 333.00 W)

nvme-pci-0d00
Adapter: PCI adapter
Composite:    +62.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +62.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +69.8°C  (low  = -273.1°C, high = +65261.8°C)
```

frankcrawford, 2025-04-27:
@amfern that is good, and shows that it will work in the future.  It does point out one other thing that that your board doesn't require any ACPI fixes for this chip.

While your fix for `sensors-detect` is fine, it isn't something I can do much with, as that is part of the whole `lm_sensors` package which is unrelated, and also they only seem to be willing to support the mainline `it87` module, not others, and I haven't seen anyway for the script to test what an module may support.

However, as a note for others, that is good, and it is good that it works for you.

amfern, 2025-04-27:
for now as a stop gap i created an aur package https://aur.archlinux.org/packages/lm_sensors-it87-git

amfern, 2025-04-28:
@frankcrawford yeah, thats a good idea
https://aur.archlinux.org/cgit/aur.git/tree/0001-Add-support-for-ITE-IT8625E-Super-IO-Sensors.patch?h=lm_sensors-it87-git

But i missed few from this PR https://github.com/lm-sensors/lm-sensors/pull/532, i will add it later

frankcrawford, 2025-10-06:
As there isn't much more I can do here at present, I'm going to close this issue.

I'm glad you got it working, and hope that the path in Arch Linux is woring fine for you.

closed 2025-10-06 by frankcrawford