frankcrawford/it87 #21: B450 AORUS PRO

JND94, 2024-04-06:
host: archlinux
sudo `./dkms-install.sh` outputs `modprobe: ERROR: could not insert 'it87': Device or resource busy`
then `sudo modprobe it87 ignore_resource_conflict=1` works but it doesn't load automatically after a reboot
(yeah `sensors` detects both chips it8792 and it8686)

How I make it load after a reboot?
Also, how to load config from Sensors config folder? I think GA-B450-AORUS-ELITE.conf fits but I don't sure

frankcrawford, 2024-04-27:
@JND94 sorry, I've been busy and missed your query.

I assume you haven't fixed the issue, so, what you need is to add a file into /etc/modprobe.d (I call it it87.conf) with the following:
```
# Local IT87 sensor options
options it87 ignore_resource_conflict=1
```

As for the GA-B450-AORUS-ELITE.conf file, if you copy it into /etc/sensors.d and it should be found.

(I will add that all this is based on the Fedora layout, but those are standard packages, so they should be correct.)

frankcrawford, 2024-06-08:
@JND94 did you get this to work?

If it does and is working okay, can you send me the output of `cat /sys/class/dmi/id/board_vendor` and `at /sys/class/dmi/id/board_name` which will allow me to automatically ignore the ACPI issue from above.

frankcrawford, 2024-06-09:
Added to it87_dmi_table in latest update.

closed 2024-06-09 by frankcrawford

Perlovka, 2025-09-29:
@frankcrawford `ignore_resource_conflict=1` works also for
```
Gigabyte Technology Co., Ltd.
B660M DS3H DDR4
```

Thank you for your work!

frankcrawford, 2025-10-01:
@Perlovka what chipset is actually in your motherboard?