frankcrawford/it87 #22: AORUS Z790 Master

daloki83, 2024-05-15:
Hi,
first of all thanks for this project! Since a few month, Linux is my main OS and so i try to get most things running like it was in WIN.  Therefore I'm fairly no expert in that kind of stuff and just try my best.

long story, short: Is it somehow possible to get my 2 missing fans detected?

from BIOS Point of view i have the following Fan ports:
CPU Fan --> AIO Fans (FAN1) @it8689-isa-0a40
CPU Opt --> Front bottom (FAN5) @it8689-isa-0a40
SysFan1 --> Back Out (FAN2) @it8689-isa-0a40
SysFan2 --> not connected
SysFan3 --> bottom back (FAN4) @it8689-isa-0a40
SysFan4 --> bottom front (FAN3) @it87952-isa-0b10
SysFan5 Pump --> AIO Pump (FAN1) @it87952-isa-0b10
SysFan6 Pump --> not connected
SysFan7 Pump --> front top **(NOT DETECTED)**
SysFan8 Pump --> front middle **(NOT DETECTED)**

watch -n1 -d 'sensors | grep fan':
```
fan1:        1854 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         378 RPM  (min =    0 RPM)
fan1:         604 RPM  (min =   10 RPM)
fan2:         467 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:         448 RPM  (min =    0 RPM)
fan5:         811 RPM  (min =    0 RPM)

```

sensors gives me:

```
it87952-isa-0b10
Adapter: ISA adapter
in0:         748.00 mV (min =  +0.00 V, max =  +2.81 V)
in1:           1.24 V  (min =  +0.00 V, max =  +2.81 V)
in2:         814.00 mV (min =  +0.00 V, max =  +2.81 V)
+3.3V:         1.43 V  (min =  +0.00 V, max =  +5.61 V)
in4:           1.18 V  (min =  +0.00 V, max =  +2.81 V)
in5:           1.83 V  (min =  +0.00 V, max =  +2.81 V)
in6:           1.42 V  (min =  +0.00 V, max =  +2.81 V)
3VSB:          3.32 V  (min =  +0.00 V, max =  +5.61 V)
Vbat:          3.08 V  
fan1:        1864 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         375 RPM  (min =    0 RPM)
temp1:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +32.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +29.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +29.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +29.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 24:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +25.0°C  (high = +80.0°C, crit = +100.0°C)
Core 32:       +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 33:       +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 34:       +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 35:       +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 36:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 37:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 38:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
[... 245 lines cut ...]
#----cut here----
If you have some drivers built into your kernel, the list above will
contain too many modules. Skip the appropriate ones!

Do you want to add these lines automatically to /etc/modules? (yes/NO)Successful!

Monitoring programs won't work until the needed modules are
loaded. You may want to run '/etc/init.d/kmod start'
to load them.

Unloading cpuid... OK

***************************************************************
Warning: the preferred way to run this script non-interactively
is with option --auto. Other methods are discouraged and may
stop working at some point in the future.
***************************************************************
```

and my /etc/modprobe.d/it87.conf
`options it87 ignore_resource_conflict=1`
i didnt use "force_id=0x8622" because i couldnt see a difference.

ahh, and my GRUB line has:
`GRUB_CMDLINE_LINUX_DEFAULT="acpi_enforce_resources=lax enabled"`

daloki83, 2024-05-18:
Not sure if this is helping, but saw that you requested for similar issues:

blacklisted and rebooted:

`isadump 0x4e 0x4f 7:`
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x4e and data register 0x4f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 95 01 40 01 00 ff 00 fe 18 02 00 8a 64 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 00 00 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 20 38 00 00 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 20 00 00 00 
c0: 01 00 00 7e 08 00 00 00 00 00 00 16 00 c0 9e 00 
d0: ff 00 00 10 00 00 00 00 00 00 00 00 00 04 00 00 
e0: 00 00 00 00 00 00 00 00 00 20 20 20 00 3e 3e 0c 
f0: 00 00 00 00 00 00 2c 00 00 00 00 00 ff 00 00 00
```
`isadump 0x2e 0x2f 7:`
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff 86 ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
`sensors-detect:`
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): y
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8695
```
[... 71 lines cut ...]
```

`isadump 0xa65 0xa66:`
```
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0xa65 and data register 0xa66.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

frankcrawford, 2024-06-05:
Thanks for all that.

I'm not sure at present what the "Gigabyte EC" is, although I have seen reports that Gigbyte are putting in some other chips to manage some things around fans.

ChirpyTurnip, 2024-09-12:
Also if it helps, here is the log from boot up:

```
Sep 12 18:16:37 Svalbard root: plugin: installing: it87-driver.plg
Sep 12 18:16:37 Svalbard root: Executing hook script: pre_plugin_checks
Sep 12 18:16:37 Svalbard root: plugin: creating: /usr/local/emhttp/plugins/it87-driver/README.md - from INLINE content
Sep 12 18:16:37 Svalbard root: plugin: running: anonymous
Sep 12 18:16:37 Svalbard root: 
Sep 12 18:16:37 Svalbard root: -----------------------IT87 Drivers found locally------------------------
Sep 12 18:16:37 Svalbard kernel: it87: it87 driver version v1.0-166-g14b1de2.20240816
Sep 12 18:16:37 Svalbard kernel: it87: Found IT8689E chip at 0xa40, revision 2
Sep 12 18:16:37 Svalbard kernel: it87: Beeping is supported
Sep 12 18:16:37 Svalbard kernel: ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20220331/utaddress-204)
Sep 12 18:16:37 Svalbard kernel: ACPI: OSL: Resource conflict; ACPI support missing from driver?
Sep 12 18:16:37 Svalbard root: 
Sep 12 18:16:37 Svalbard root: ----------------Installation of IT87 Drivers successful------------------
Sep 12 18:16:37 Svalbard root: 
Sep 12 18:16:37 Svalbard root: plugin: it87-driver.plg installed
Sep 12 18:16:37 Svalbard root: plugin: it87-driver.plg installed
```

frankcrawford, 2024-09-22:
@ChirpyTurnip Looking at the `pwmconfig` script, it really doesn't know what fan is associated with which PWM, rather, it changes the PWM settings and then checks to see which fan changes at the same time.  Unfortunately, this means that when it is not really working it just gets random associations.

The problem is almost certainly the issue listed in issue #11 where Gigabyte have introduced a new chip into the mix and currently none of the drivers handle it.  If you read through that issue you will see some workarounds you can try.

Also, as a minor fix, I've added you board into the default list to ignore ACPI support, so you no longer need to add `ignore_resource_conflict=1` to the command line.

frankcrawford, 2024-09-22:
@daloki83 if you are still following this, see the comment above, but also, can you send me the details you get from `cat /sys/class/dmi/id/board_vendor` and `/sys/class/dmi/id/board_name` just so I can track what chips are in which boards.

LittleLightLittleFire, 2024-12-26:
Not him but I have the ELITE board and same chip (8689E), fan sensors work but no control.

```
$ cat /sys/class/dmi/id/board_vendor
Gigabyte Technology Co., Ltd.
```
```
$ cat /sys/class/dmi/id/board_name
Z690 AORUS ELITE DDR4
```
```
$ isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 89 01 00 2c 20 73 00 01 80 c0 32 88 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 00 00 00 00 
70: 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 c0 00 00 00 00 
c0: 01 40 00 41 00 ff ff ff 01 00 00 00 00 20 00 02 
d0: c9 00 20 22 00 fd ff ff 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 20 12 49 00 00 0a 7c 
f0: 00 00 00 00 00 00 0d 00 00 d0 49 60 00 9f 00 0f 
```
```
$ isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```
Can compile/test, let me know!

daloki83, 2025-01-04:
> @daloki83 if you are still following this, see the comment above, but also, can you send me the details you get from `cat /sys/class/dmi/id/board_vendor` and `/sys/class/dmi/id/board_name` just so I can track what chips are in which boards.

sorry for the late reply 

```
cat /sys/class/dmi/id/board_vendor
Gigabyte Technology Co., Ltd.
```

```
cat /sys/class/dmi/id/board_name
Z790 AORUS MASTER
```

frankcrawford, 2025-01-11:
@daloki83 I'll set it up to automatically override ACPI issues.

frankcrawford, 2025-03-23:
If you have any further follow up for this, feel free to reopen or open a new issue.

closed 2025-03-23 by frankcrawford