frankcrawford/it87 #26: Report for B650M GAMING X AX (rev 1.3)

emansom, 2024-06-04:
Per instructions in the README, i'm providing information to hopefully get fan control on the B650M GAMING X AX (rev 1.3) working.

Output of commands:

<details>
  <summary>sensors-detect</summary>
  
  ```shell
  09:45 ewout@am5ws ~% sudo sensors-detect                                           
  # sensors-detect version 3.6.0+git
  # System: Gigabyte Technology Co., Ltd. B650M GAMING X AX [Default string]
  # Kernel: 6.9.3-1-cachyos x86_64
  # Processor: AMD Ryzen 5 7600 6-Core Processor (25/97/2)
  
  This program will help you determine which kernel modules you need
  to load to use lm_sensors most effectively. It is generally safe
  and recommended to accept the default answers to all questions,
  unless you know what you're doing.
  
  Some south bridges, CPUs or memory controllers contain embedded sensors.
  Do you want to scan for them? This is totally safe. (YES/no): 
  Silicon Integrated Systems SIS5595...                       No
  VIA VT82C686 Integrated Sensors...                          No
  VIA VT8231 Integrated Sensors...                            No
  AMD K8 thermal sensors...                                   No
  AMD Family 10h thermal sensors...                           No
  AMD Family 11h thermal sensors...                           No
  AMD Family 12h and 14h thermal sensors...                   No
  AMD Family 15h thermal sensors...                           No
  AMD Family 16h thermal sensors...                           No
  AMD Family 17h thermal sensors...                           No
  AMD Family 15h power sensors...                             No
  AMD Family 16h power sensors...                             No
  Hygon Family 18h thermal sensors...                         No
  AMD Family 19h thermal sensors...                           No
  Intel digital thermal sensor...                             No
  Intel AMB FB-DIMM thermal sensor...                         No
  Intel 5500/5520/X58 thermal sensor...                       No
  VIA C7 thermal sensor...                                    No
  VIA Nano thermal sensor...                                  No
  
  Some Super I/O chips contain embedded sensors. We have to write to
  standard I/O ports to probe them. This is usually safe.
  Do you want to scan for Super I/O sensors? (YES/no): 
  Probing for Super-I/O at 0x2e/0x2f
  Trying family `National Semiconductor/ITE'...               No
  Trying family `SMSC'...                                     No
  Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
  Trying family `ITE'...                                      Yes
  Found unknown chip with ID 0x8689
  Probing for Super-I/O at 0x4e/0x4f
  Trying family `National Semiconductor/ITE'...               No
  Trying family `SMSC'...                                     No
  Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
  Trying family `ITE'...                                      No
  
  Some systems (mainly servers) implement IPMI, a set of common interfaces
  through which system health data may be retrieved, amongst other things.
  We first try to get the information from SMBIOS. If we don't find it
  there, we have to read from arbitrary I/O ports to probe for such
  interfaces. This is normally safe. Do you want to scan for IPMI
  interfaces? (YES/no): ^C
  09:53 ewout@am5ws ~% 
  ```
</details>

<details>
  <summary>sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7</summary>
  
  ```shell
  09:48 ewout@am5ws ~% sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
  [sudo] wachtwoord voor ewout: 
  WARNING! Running this program can cause system crashes, data loss and worse!
  I will probe address register 0x2e and data register 0x2f.
  Probing bank 7 using bank register 0x07.
  Continue? [Y/n] 
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
  00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
  10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  20: 86 89 02 00 2e 20 b3 00 00 00 80 30 88 40 00 00 
  30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  60: 00 00 0a 00 20 00 00 00 00 00 00 00 00 00 00 00 
  70: 00 00 20 00 00 17 00 00 00 00 00 00 00 00 00 00 
  80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
  b0: 00 00 00 00 00 00 00 00 00 00 00 c0 00 00 00 00 
  c0: 01 00 00 40 00 ff ff ff 01 00 00 00 00 68 00 00 
  d0: b5 00 08 0a 00 fd ff ff 00 00 00 00 00 00 00 00 
  e0: 00 00 00 00 00 00 00 50 00 20 12 49 00 00 0a 7c 
  f0: 00 00 00 00 00 00 0d 00 00 d0 49 60 00 6e 00 ff 
  09:49 ewout@am5ws ~% 
  ```
</details>

Can't determine the device address to run the third command, as it's missing from the output of `sensors-detect`.

emansom, 2024-06-04:
Motherboard says it's an `IT8689E`:

[image]

frankcrawford, 2024-06-05:
Thanks for that.  I'll look at it in the next few days.

I think that chip is already supported, so is the issue you are seeing that the driver won't load?  I guess I'll check through the attachments.

Have you attached them, as I can't see them here?

emansom, 2024-06-05:
> I think that chip is already supported, so is the issue you are seeing that the driver won't load? I guess I'll check through the attachments.

Yes, it won't load automatically.

A dmi match like done for other motherboards might work looking at code, though not sure.
Will test with patching.

> Have you attached them, as I can't see them here?

I wrapped them in a `</details>` block, simply click on them.

emansom, 2024-06-05:
Digging further, it seems I was testing with the upstream/built-in version of this module inside the kernel (CachyOS kernel).

Now building a kernel without the built-in module compiled in, so I can test a DKMS built one.

emansom, 2024-06-05:
Any plans on submitting patches to LKML to update the upstream/built-in it87 driver?

emansom, 2024-06-05:
Testing the DKMS built one (latest git, using AUR package [it87-dkms-git](https://aur.archlinux.org/packages/it87-dkms-git)) results in the following:

<details>
  <summary>sudo modprobe it87</summary>
  
  ```shell
  00:13 ewout@am5ws ~% sudo modprobe it87
  [sudo] wachtwoord voor ewout:
  modprobe: ERROR: could not insert 'it87': Device or resource busy
  00:13 ewout@z97x ~% sudo modprobe it87 force_id=0x8689
  modprobe: ERROR: could not insert 'it87': Device or resource busy
  ```
</details>

frankcrawford, 2024-06-06:
@emansom have you rebooted the system since the DKMS rebuild?  While it moves the original module out of the way, it doesn't always manage to unload it cleanly.

The error you are seeing sounds like the LKML version is still loaded.

If you don't want to reboot you could try `rmmod it87` followed by `modprobe it87`

As for upstreaming it, yes, that is going on, but every difference needs to be justified, so it is a very slow process, especially as I didn't write lots of those differences, and need to go through them very carefully.

frankcrawford, 2024-06-06:
Oh, looking at the code a bit further, currently that board doesn't automatically ignore ACPI issues.  Can you check `dmesg` and see if you have a line like

```
ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20230628/utaddress-204)
```

If you do see that, add the option `ignore_resource_conflict=1`, i.e. run `modprobe it87 ignore_resource_conflict=1` and let me know if that works.  If it does, you can temporarily add that option into /etc/modprobe.d/it87.conf (or equivalent).

I can then get you to do a few tests to see if there are any issues making it a permanent fix.

emansom, 2024-06-06:
> @emansom have you rebooted the system since the DKMS rebuild? While it moves the original module out of the way, it doesn't always manage to unload it cleanly.

Yes. I rebooted after I recompiled the kernel with the built-in module disabled.

> The error you are seeing sounds like the LKML version is still loaded.

No module is loaded. As you can see from the output of `lsmod`.

> As for upstreaming it, yes, that is going on, but every difference needs to be justified, so it is a very slow process, especially as I didn't write lots of those differences, and need to go through them very carefully.

Yeah they seem to be stuck in the 80s still. Appreciate it that you're going down that rabbit hole!

emansom, 2024-06-06:
Output of `lshw`:

There are three versions of this exact motherboard. This motherboard is a rev 1.3 model.
I'm not entirely sure if the other revisions have different ITE chips and have no way to verify this myself.
I also see no easy way to identify the revision of this board from DMI/ACPI info.
<details>
  <summary>sudo lshw</summary>
  
  ```shell
  18:05 ewout@am5ws ~% sudo lshw
  am5ws
      description: Desktop Computer
      product: B650M GAMING X AX (Default string)
      vendor: Gigabyte Technology Co., Ltd.
      version: Default string
      serial: Default string
      width: 64 bits
      capabilities: smbios-3.6.0 dmi-3.6.0 smp vsyscall32
      configuration: boot=normal chassis=desktop family=B650 MB sku=Default string uuid=03560274-043c-05f9-b806-b60700080009
    *-core
         description: Motherboard
         product: B650M GAMING X AX
         vendor: Gigabyte Technology Co., Ltd.
         physical id: 0
         version: x.x
         serial: Default string
         slot: Default string
       *-firmware
            description: BIOS
            vendor: American Megatrends International, LLC.
            physical id: 0
            version: FA5a
            date: 04/12/2024
            size: 64KiB
            capacity: 32MiB
            capabilities: pci upgrade shadowing cdboot bootselect socketedrom edd int13floppynec int13floppytoshiba int13floppy360 int13floppy1200 int13floppy720 int13floppy2880 int5printscreen int9keyboard int14serial int17printer int10video acpi usb biosbootspecification uefi
       *-cache:0
            description: L1 cache
            physical id: b
            slot: L1 - Cache
            size: 384KiB
            capacity: 384KiB
            clock: 1GHz (1.0ns)
            capabilities: pipeline-burst internal write-back unified
            configuration: level=1
       *-cache:1
            description: L2 cache
            physical id: c
            slot: L2 - Cache
            size: 6MiB
            capacity: 6MiB
            clock: 1GHz (1.0ns)
            capabilities: pipeline-burst internal write-back unified
            configuration: level=2
       *-cache:2
            description: L3 cache
            physical id: d
            slot: L3 - Cache
            size: 32MiB
            capacity: 32MiB
            clock: 1GHz (1.0ns)
            capabilities: pipeline-burst internal write-back unified
            configuration: level=3
       *-cpu
            description: CPU
            product: AMD Ryzen 5 7600 6-Core Processor
            vendor: Advanced Micro Devices [AMD]
            physical id: e
            bus info: cpu@0
            version: 25.97.2
[... 973 lines cut ...]
         logical name: input1
         logical name: /dev/input/event1
         capabilities: platform
    *-input:2
         product: Video Bus
         physical id: 3
         logical name: input2
         logical name: /dev/input/event2
         capabilities: platform
    *-input:3
         product: solaar-keyboard
         physical id: 4
         logical name: input33
         logical name: /dev/input/event22
         logical name: /dev/input/mouse1
         capabilities: usb
    *-input:4
         product: WI-C100 (AVRCP)
         physical id: 5
         logical name: input42
         logical name: /dev/input/event23
         capabilities: bluetooth
  18:05 ewout@am5ws ~% 
  ```
</details>

emansom, 2024-06-06:
I think `gigabyte_wmi` might be the cause of the conflict, not sure.

frankcrawford, 2024-06-07:
@emansom the actual conflict is basically a stupid setup by Gigabyte, as they have claimed the address space, but often don't use it.  In this case they are almost correct, as `gigabyte_wmi` is the "official way" to look up the info, but misses most information.  For many boards `gigabyte_wmi` doesn't work, but they still have the address space claimed.

Anyway, that is just an aside, the possible "multiple" access to the same memory region is a risk, but I am yet to have a report that it actually caused an issue.  However, can you do some further and monitoring for a couple of days, to see if it is all okay, and then send me some additional items if you think it should be automatically enabled.

What I need is `cat /sys/class/dmi/id/board_vendor` and `cat /sys/class/dmi/id/board_name` for the hardware match.

emansom, 2024-06-07:
> @emansom the actual conflict is basically a stupid setup by Gigabyte, as they have claimed the address space, but often don't use it. In this case they are almost correct, as `gigabyte_wmi` is the [...]

Would blacklisting/not compiling the `gigabyte_wmi` module be a safer option?

> Anyway, that is just an aside, the possible "multiple" access to the same memory region is a risk, but I am yet to have a report that it actually caused an issue. However, can you do some further [...]

I tried yesterday to control the PWM of the fans via [`fan2go`](https://github.com/markusressel/fan2go) with both modules active (`gigabyte_wmi` loaded and this fork `it87` via DKMS active with `ignore_resource_conflict=1`). Firstly it calibrated to all the wrong min and max PWM values, and even when triggering manually via the regular hwmon paths: it didn't change the RPM of the fans in the slightest.

> What I need is `cat /sys/class/dmi/id/board_vendor` and `cat /sys/class/dmi/id/board_name` for the hardware match.

<details>
  <summary>cat /sys/class/dmi/id/board_vendor</summary>
  
  ```shell
  19:47 root@am5ws ~# cat /sys/class/dmi/id/board_vendor                                   
  Gigabyte Technology Co., Ltd.
  19:47 root@am5ws ~# 
  ```
</details>

<details>
  <summary>cat /sys/class/dmi/id/board_name</summary>
  
  ```shell
  19:48 root@am5ws ~# cat /sys/class/dmi/id/board_name
  B650M GAMING X AX
  19:48 root@am5ws ~#
  ```
</details>

emansom, 2024-06-07:
Further digging. This is weird:

```shell
22:46 root@am5ws ~# echo 1 > /sys/devices/platform/it87.2624/hwmon/hwmon5/pwm1_enable
22:47 root@am5ws ~# cat /sys/devices/platform/it87.2624/hwmon/hwmon5/pwm1_enable     
0
22:47 root@am5ws ~# 
```

frankcrawford, 2024-06-08:
@emansom unfortunately `gigabyte_wmi` doesn't really affect whether the BIOS claims the addresses.  You can blacklist it if you wish, but that does not affect the error, nor how the BIOS accesses the ports.

Now, as for the fan management, currently it will probably not work, as Gigabyte have done something strange.  You can check issue #11 for more details.  Another project does seem to have a fix, which I am looking at, and will see what I can do about it in the near future.  In the meantime you can try the work-around given in the issue.

frankcrawford, 2024-06-08:
I've added this model to automatically set `ignore_resource_conflict=1` in the latest commit.

emansom, 2024-06-09:
Do you by chance know any USB (preferably header based) or PCI-e based PWM controllers available as consumer products that have good quality mainline kernel drivers?

frankcrawford, 2024-06-09:
Unfortunately no.

closed 2024-09-22 by frankcrawford

emansom, 2024-10-10:
> Unfortunately no.

@frankcrawford Came across this, will buy the Micro version and test it with a Noctua fan hub (NA-FH1) next month.

It seems like the most viable solution in the mean time while Gigabyte fan support is suboptimal.

I'll report back if it works and if it does: consider adding a entry to the README and project docs to point users to this open-source USB based PWM solution instead.

https://github.com/SasaKaranovic/OpenFanController

https://shop.sasakaranovic.com/products/openfan-pc-fan-controller

https://shop.sasakaranovic.com/products/openfan-micro

@SasaKaranovic