frankcrawford/it87 #29: ITE8689: No such device

Forcii1, 2024-07-17:
Mobo: B650 Eagle AX
IO-Chip: ITE 8689E

What I did:
git clone https://github.com/frankcrawford/it87
cd it87
```
make clean
make
sudo make install
sudo modprobe it87 ignore_resource_conflict=1 force_id=0x8689
```
Everything works but then I get the No such device error.
Don't know if this is important but  `sudo modprobe it87 ignore_resource_conflict=1 force_id=0x8628` works

frankcrawford, 2024-08-15:
@Forcii1 you say that it is an ITE 8689E chip, but how have you confirmed that?  If you are using this driver then it would/should automatically detect a chip with ID 0x8689.

So, firstly, if you leave off the `force_id=0x8689` option, what is the result?

Secondly, what does `sensors-detect` show for possible chips on your motherboard?

frankcrawford, 2024-08-16:
I was thinking about this last night and am wondering which it87 module you have loaded?  The one from this repo, or the one from the kernel mainline?

Can you do `modinfo it87` and see what it reports?  Alternatively do `dmesg | grep it87` and see if it reports a version number.

Forcii1, 2024-08-24:
`dmesg | grep it87` just shows nothing
`modinfo it87` 
/usr/lib/modules/6.10.5-1-default/kernel/drivers/hwmon/it87.ko.zst
license:        GPL
description:    IT8705F/IT871xF/IT872xF hardware monitoring driver
author:         Chris Gauthron, Jean Delvare <jdelvare@suse.de>
suserelease:    openSUSE Tumbleweed
srcversion:     BC9B0FE462169CBCF80ACFF
alias:          dmi*:rvn*nVIDIA*:rn*FN68PT*:
depends:        hwmon-vid
retpoline:      Y
intree:         Y
name:           it87
vermagic:       6.10.5-1-default SMP preempt mod_unload modversions 
sig_id:         PKCS#7
signer:         openSUSE Secure Boot CA
sig_key:        FA:BE:D8:BF:40:9A:5E:65
sig_hashalgo:   sha256
signature:      85:B4:1B:76:79:E1:7D:9D:C7:94:34:41:D8:54:3D:3E:6F:35:B3:4E:
                07:35:29:27:8E:E8:30:9C:20:3C:AC:F9:A9:40:8C:ED:45:56:DC:B3:
                68:3A:B0:41:DE:89:42:2B:3D:63:5C:A6:BC:A9:CA:4B:8F:B4:20:25:
                B7:1A:F0:79:BC:CA:C2:87:6A:DC:A0:71:D2:52:D1:B3:8F:64:0F:D4:
                02:A4:E8:63:1A:28:0B:00:96:B1:F3:7C:DE:5D:F7:3B:CB:3C:5B:55:
                76:21:FF:4D:1F:DD:9F:44:53:D7:95:CF:97:9E:03:5B:8F:D8:75:D3:
                36:0A:19:CD:D1:49:3E:1E:19:36:EA:7B:24:41:AE:76:39:9B:1E:3F:
                10:FD:C5:9E:BC:AA:71:16:AE:A8:63:F6:64:BB:89:F2:4C:4F:97:2E:
                8A:20:C5:C0:AE:CB:C3:F8:18:A8:5B:26:67:10:1A:B9:32:88:90:25:
                CC:F3:DD:0F:13:28:E3:4F:0C:3C:70:CA:C2:F5:DD:AE:7E:8D:4F:7F:
                0B:4F:2D:59:D2:C1:32:41:A1:9A:A5:93:F7:6F:25:5A:6E:AA:8F:95:
                AA:B1:BC:89:6B:95:30:A1:99:72:66:35:41:ED:AD:32:C1:99:92:BC:
                58:05:30:4B:75:0F:71:81:09:E2:02:E8:EC:DD:3C:74
parm:           force_id:Override one or more detected device ID(s) (array of ushort)
parm:           ignore_resource_conflict:Ignore ACPI resource conflict (bool)
parm:           update_vbat:Update vbat if set else return powerup value (bool)
parm:           fix_pwm_polarity:Force PWM polarity to active high (DANGEROUS) (bool)

And here has to be the error: `description:    IT8705F/IT871xF/IT872xF hardware monitoring driver`
I need IT87xE
How do I change the driver?

Forcii1, 2024-08-24:
So I went into `/lib/modules/6.10.5-1-default/kernel/drivers/hwmon`
`sudo make install` creates it87.ko. Opensuse is loading it87.ko.zst. So I changed it87.ko to it.ko.zst.

`sudo modprobe it87 ignore_resource_conflict=1 force_id=0x8689` works now. 
`sensors-detect` gives me this output: 
`Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689
Probing for Super-I/O at 0x4e/0x4f
`
`sensors` does not show more stuff.

`dmesg | grep it87` now has following output: 
`[  607.728387] [   T9198] it87: loading out-of-tree module taints kernel.
[  607.728393] [   T9198] it87: module verification failed: signature and/or required key missing - tainting kernel
[  607.729108] [   T9198] it87: it87 driver version v1.0-163-ga828995.20240609
[  607.729211] [   T9198] it87: Found IT8689E chip at 0xa40, revision 2
[  607.729269] [   T9198] it87: Beeping is supported
`

frankcrawford, 2024-08-25:
@Forcii1 firstly, `sensors-detect` only knows about chipsets in the mainline kernel module, so knows nothing about the ITE8689E, and to make it worse, it unloads modules, so will screw up any following `sensors` usage.  However, it does show that the chip does exist on your motherboard.

So, instead, just do a `sudo modprobe it87 ignore_resource_conflict=1` (it should detect the ITE8689E automatically). and then run `sensors` and see what you see now.

Following this, you need to add in somewhere like `/etc/modprobe.d/it87.conf` the following:
```
# Local IT87 sensor options
options it87 ignore_resource_conflict=1
```

As for it now being found without compress, firstly, check if the module exists in `/usr/lib/modules/6.10.5-1-default/kernel/drivers/hwmon/it87.ko.zst`, if so, the installation is not working correctly, as it should have removed that, as well why it didn't compress.  That is probably something to do with SuSE's kernel installation process.

frankcrawford, 2025-03-23:
If there is any further follow up for this, feel free to reopen or open a new issue.

closed 2025-03-23 by frankcrawford