frankcrawford/it87 #33: installation fail kernel 6.8.0-41

dga4130-renato, 2024-09-07:
kernel 6.8.0-41 ubuntu 24.04

sudo make dkms
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
make -j20 KERNELRELEASE=6.8.0-41-generic TARGET=6.8.0-41-generic...
Signing module /var/lib/dkms/it87/unknown/build/it87.ko
Cleaning build area...

it87.ko.zst:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.8.0-41-generic/updates/dkms/
depmod...
modprobe: ERROR: could not insert 'it87': Device or resource busy
make: *** [Makefile:102: dkms] Error 1

dga4130-renato, 2024-09-07:
cat /var/lib/dkms/it87-master/.20240907/build/make.log
DKMS make.log for it87-master-.20240907 for kernel 6.8.0-41-generic (x86_64)
sab 7 set 2024, 16:57:01, CEST
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
make[3]: ***  Nessuna regola per generare l'obiettivo «/var/lib/dkms/it87-master/.20240907/build/it87-master.o», necessario per «/var/lib/dkms/it87-master/.20240907/build/».  Arresto.
make[2]: *** [Makefile:1926: /var/lib/dkms/it87-master/.20240907/build] Errore 2
make[1]: *** [Makefile:73: modules] Errore 2

dga4130-renato, 2024-09-07:
partially solved this way:
sudo make -f Makefile DRIVER=it87 DRIVER_VERSION=3.0-20240908 DKMS_ROOT_PATH=/usr/src/it87-3.0-20240908 dkms
sudo nano /var/lib/dkms/it87
options it87 ignore_resource_conflict=1
save
sudo nano /etc/modules
it87
save

i think there is something else doing with hooks, but not sure. anyway works.
P.s. 
sudo apt install gcc-14 for last kernels

hope this will help someone

frankcrawford, 2024-09-18:
@dga4130-renato there are two ways to fix this, one in the code, and one by you, although both will need some details from you.

However first off, in my case `/var/lib/dkms/it87` is a directory, not a file you can edit.  What Linux distribution are you using?

Anyway, the more standard fix is similar and is
```
sudo nano /etc/modprobe.d/it87.conf
options it87 ignore_resource_conflict=1
save
```
(i.e. same info, just different location).

The way I can get it fixed in the code is if you supply some information about the board you are using and then I can automatically set that option.  To do this run the following and supply me with the results:
```
sudo cat /sys/class/dmi/id/board_vendor
sudo cat /sys/class/dmi/id/board_name
```

dga4130-renato, 2024-09-19:
"However first off, in my case /var/lib/dkms/it87 is a directory, not a file you can edit. What Linux distribution are you using?"

yes, you are correct... i made a mistake here " /var/lib/dkms/it87

Details:

UBUNTU 24.04.1 LTS
Kernel 6.11.0

Gigabyte Technology Co., Ltd.
H610M S2H V2

sudo nano /etc/modprobe.d/it87.conf
options it87 force_id=0x8689 ignore_resource_conflict=1
save

sudo sensors-detect 
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689

also i saw this in the code.. it87.c

	IT87_DMI_MATCH_GBT("H610M H DDR4", it87_dmi_cb,
			   &it87_acpi_ignore),
		/* IT8689E */
	IT87_DMI_MATCH_GBT("H610M S2H V2", it87_dmi_cb,
			   &it87_acpi_ignore),

so i added the lower line...IT87_DMI_MATCH_GBT("H610M S2H V2", it87_dmi_cb,

maybe this is not the correct way to do it, or maybe i was wrong ... but seems to work ok ...

frankcrawford, 2024-09-19:
@dga4130-renato your change is pretty much what I'm going to add, so yes it is right.  I may add a comment, etc.

dga4130-renato, 2024-09-19:
Nice ! Thanks for spending your time here 👍

nferch, 2024-12-02:
Same issue for me, same successful workaround.

```
Gigabyte Technology Co., Ltd.
Q370M D3H GSM PLUS
```

kernel log:

`ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20190816/utaddress-204)`

```
it8686-isa-0a40                                                                                                                     Adapter: ISA adapter                                                                                                                in0:         636.00 mV (min =  +0.00 V, max =  +3.06 V)                                                                             in1:           2.02 V  (min =  +0.00 V, max =  +3.06 V)                                                                             in2:           2.02 V  (min =  +0.00 V, max =  +3.06 V)                                                                             in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)                                                                             in4:           0.00 V  (min =  +0.00 V, max =  +3.06 V)  ALARM                                                                      in5:           1.06 V  (min =  +0.00 V, max =  +3.06 V)                                                                             in6:           1.20 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.34 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          2.78 V
fan1:         799 RPM  (min =   10 RPM)
fan2:        1256 RPM  (min =   10 RPM)
temp1:        +32.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermistor
temp2:        +45.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermistor
temp3:        +31.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = Intel PECI
temp4:         +0.0°C  (low  =  +0.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +34.0°C  (low  =  +0.0°C, high = +60.0°C)  sensor = thermistor
temp6:         +0.0°C  (low  =  +0.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

frankcrawford, 2025-03-23:
Added the `Q370M D3H GSM PLUS` to the DMI table.  Sorry about the long delay.

closed 2025-03-23 by frankcrawford