frankcrawford/it87 #34: IT87_REG_CHIPID

dga4130-renato, 2024-09-22:
Hi,
I notice that the beep command does not work correctly, if the bios beeps, or the grub beep works fine , but after kernel loading the pcspkr just makes a click. 

is this the blame ? #define IT87_REG_CHIPID       0x58

frankcrawford, 2024-11-01:
@dga4130-renato not sure.  I know it works fine in older chips, but haven't looked at it too closely lately.

Also, I doubt the setting would actually make a big difference to the sound, itself, as it would be hardware driven, so I think it would either work or not work.

What chip do you have?

Frank

dga4130-renato, 2024-11-01:
0x8689

frankcrawford, 2025-03-23:
I'm closing this off, if you have wish, feel free to reopen or open a new issue.

closed 2025-03-23 by frankcrawford