frankcrawford/it87 #41: sensors-detect does not detect driver

thegame4craft, 2025-01-05:
This driver for my SuperIO Chip (It8665e) is not detected my sensors-detect:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): <Enter>
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8665E Super IO Sensors'                        Success!
    (address 0x290, driver `to-be-written')                                       <--- This
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
```

```
Now follows a summary of the probes I have just done.
Just press ENTER to continue:

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Driver `to-be-written':
  * ISA bus, address 0x290
    Chip `ITE IT8665E Super IO Sensors' (confidence: 9)

Note: there is no driver for ITE IT8665E Super IO Sensors yet.
Check https://hwmon.wiki.kernel.org/device_support_status for updates.
```

__System Information__:
sensors-detect version 3.6.0+git
Board: ASUSTeK COMPUTER INC. PRIME B450-PLUS
Kernel: 6.12.7-arch1-1 x86_64
Processor: AMD Ryzen 5 3600 6-Core Processor (23/113/0)
Yay-Package: local/it87-dkms-git 269.213db3a-1
driver it87 loaded using modprobe and verified via `lsmod | grep it87`

__Output as specified in Readme__:
```
[robin@PC-RobinLinux ~]$ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 86 65 06 40 27 91 ff 01 00 11 00 49 85 c8 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
60: 00 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 01 00 38 00 10 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0: 91 ff 01 00 11 00 00 00 00 20 00 00 10 00 00 20
d0: ff 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00
e0: 00 00 00 00 00 00 00 10 00 07 07 07 00 00 40 0c
f0: 00 00 00 00 00 00 0e 00 2c 02 00 00 00 00 00 00
[robin@PC-RobinLinux ~]$
[robin@PC-RobinLinux ~]$
[robin@PC-RobinLinux ~]$ sudo isadump 0x295 0x296
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x295 and data register 0x296.
Continue? [Y/n]
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 09 01 7f ab 00 00 40 00 00 c0 44 08 10 7f ff c2
10: 41 e5 00 00 7f 00 00 00 7f 00 00 00 7f 1a 00 00
20: 7f 00 20 00 7f 00 20 00 7f 00 20 00 7f 30 c0 9e
30: 7f 00 20 00 7f 00 20 00 7f 00 20 00 7f 01 01 01
40: 7f 00 20 00 7f 00 20 00 7f 00 20 00 7f 00 00 00
50: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c
60: 7f 00 00 00 7f 00 00 00 7f 00 00 00 7f 3c 3c 3c
70: 08 1e 00 4c 4b 00 00 00 00 00 46 99 14 00 0f a8
80: ca 02 00 00 00 00 00 00 00 00 00 00 00 06 00 23
90: 00 00 00 ff ff 00 00 ff 40 00 00 00 00 61 00 83
a0: 7f 7f 7f 80 00 00 0f 80 80 7f 7f ff 00 00 0f ff
b0: d8 0e 00 00 77 cc bb fc ff ff ff ff ff ff ff ff
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[robin@PC-RobinLinux ~]$
```

(If any information is missing, please let me know so i can add it)

frankcrawford, 2025-01-11:
First `sensors-detect` is actually independent of any of the kernel modules, although is roughly aligned with the mainline kernel and knows nothing about the additional chips this driver covers.

However, more importantly to track this issue, can you run `modinfo it87` and see what it returns, as I want to confirm it is finding the updated one and not the one in the mainline kernel.

thegame4craft, 2025-01-19:
modinfo it87 as requested:

```
filename:       /lib/modules/6.12.8-arch1-1/updates/dkms/it87.ko.zst
version:        v1.0-169-g213db3a.20240922
license:        GPL
description:    IT87xxF/IT86xxE hardware monitoring driver
author:         Chris Gauthron, Jean Delvare <jdelvare@suse.de>, Frank Crawford
srcversion:     5FDE6BF1CE20797221F4AE5
alias:          dmi*:rvn*nVIDIA*:rn*FN68PT*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*Z790AORUSMASTER*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*Z790AORUSELITEAX*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*Z690AORUSPRO*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*Z690AORUSPRODDR4*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*H610MS2HV2*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*H610MHDDR4*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X670EAORUSMASTER*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X670AORUSELITEAX*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*B650MGAMINGXAX*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X570SAEROG*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X570IAORUSPROWIFI*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X570AORUSULTRA*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X570AORUSPROWIFI*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X570AORUSPRO*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X570AORUSMASTER*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X570AORUSELITEWIFI*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*X570AORUSELITE*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*B560IAORUSPROAX*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*B550AORUSPROAC*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*Z490AORUSELITEAC*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*B450AORUSPRO-CF*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*Z390AORUSULTRA-CF*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*TRX40AORUSXTREME*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*Z97X-GamingG1*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*A520IAC*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*AX370*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*AB350*:
alias:          dmi*:rvn*GigabyteTechnologyCo.,Ltd.*:rn*A320M-S2HV2-CF*:
depends:        hwmon-vid
name:           it87
retpoline:      Y
vermagic:       6.12.8-arch1-1 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         DKMS module signing key
sig_key:        7A:25:54:D3:CC:86:47:03:D2:76:F4:4C:C8:A3:D3:CA:B8:32:B6:C7
sig_hashalgo:   sha512
signature:      39:C8:B9:12:05:70:C3:29:A6:86:9F:AE:4A:11:A2:9E:61:FF:C7:A9:
		65:73:0E:E9:4F:35:FA:FD:C5:FE:20:4B:4F:33:DC:83:58:31:A0:6E:
		75:33:67:B2:24:29:1D:2C:DA:AC:A3:7B:FE:FF:3D:AF:B1:F2:73:3D:
		B9:FA:D8:55:57:58:C2:2C:CD:5E:A9:9A:40:31:36:E3:F8:71:7E:94:
		73:64:CB:56:4F:EF:56:E4:EB:1F:B6:37:EF:10:A5:78:3E:70:F7:49:
		66:40:8C:DD:ED:BE:56:A6:0E:46:38:7A:97:75:5F:21:D5:A1:7F:F9:
		00:21:08:74:62:1D:74:A5:58:BE:44:2E:2A:E2:57:98:95:E7:00:3C:
		7C:E4:98:EC:B0:5B:78:9A:C0:32:37:CC:69:9A:C5:6D:F9:6F:82:FF:
		78:19:49:2F:BC:E9:71:C6:27:96:16:C8:E7:7F:D5:91:ED:1E:9B:E1:
		E5:B5:44:DA:71:76:2B:E6:09:34:5A:B7:97:42:42:D6:B7:B6:53:F4:
		D3:6F:44:CE:34:99:70:96:37:BC:6A:E5:A8:07:2E:0C:12:98:72:2B:
		0B:4D:C9:F0:31:FA:6E:48:33:8A:9C:96:A0:DA:30:6C:63:9E:80:25:
		06:FF:D3:64:43:EB:40:BD:66:21:E3:6E:1D:0E:1E:32
parm:           force_id:Override one or more detected device ID(s) (array of ushort)
parm:           ignore_resource_conflict:Ignore ACPI resource conflict (bool)
parm:           mmio:Use MMIO if available (bool)
parm:           update_vbat:Update vbat if set else return powerup value (bool)
parm:           fix_pwm_polarity:Force PWM polarity to active high (DANGEROUS) (bool)

```

closed 2025-01-27 by thegame4craft