frankcrawford/it87 #42: IT87 module in 6.8.12 not working anymore

VonKossa, 2025-01-11:
Hi!
I  have been running Linux 5.10 with IT87 module without problems, it is an old gigabyte motherboard with a it8728-290 chipset from sandybridge era. 
Upgraded to Linux 6.12.8 yesterday and now the module is not detected anymore, i cannot see it listed at all , I am compiling from source so i am certain the module is compiled with the kernel.
Looked on kernel.org and could see that the following changes have been pushed by you since 5.10.230 which worked without problems, do you have any clue why this would occure with it8728?

2024-04-30	hwmon: (it87) Remove tests nolonger required
2024-04-30	hwmon: (it87) Test for chipset before entering configuration mode
2024-04-30	hwmon: (it87) Do not enter configuration mode for some chiptypes
2024-04-30	hwmon: (it87) Rename FEAT_CONF_NOEXIT to FEAT_NOCONF as more descriptive of r...
2023-08-21	hwmon: (it87) Add support to detect sensor type AMDTSI
2023-08-21	hwmon: (it87) Improve temperature reporting support
2023-08-21	hwmon: (it87) Split temperature sensor detection to separate function
2023-06-08	hwmon: (it87) Generalise support for FAN_CTL ON/OFF
2023-06-08	hwmon: (it87) Allow for chips with only 4 temp sensors
2023-06-08	hwmon: (it87) Update IT8732F chip for 4 fans and PWMs
2023-06-08	hwmon: (it87) Add controls for chips with only 4 PWMs
2023-06-08	hwmon: (it87) Add controls for chips with only 4 fans
2023-04-21	hwmon: (it87) Disable/enable SMBus access for IT8622E chipset
2023-04-21	hwmon: (it87) Add calls to smbus_enable/smbus_disable as required
2023-04-21	hwmon: (it87) Test for error in it87_update_device
2023-04-19	hwmon: (it87) Disable SMBus access for environmental controller registers.
2023-04-19	hwmon: (it87) Use voltage scaling macro where appropriate

frankcrawford, 2025-01-11:
@VonKossa no, I can't see any particular reason for it to fail.  It is still in the current kernel module, as you noted, so it is likely to be something with the detection.

Can you do two things, firstly run `lsmod | grep it87` to see if it is loaded (probably won't be), and secondly, run `sensors-detect` to see if it finds the chip.

VonKossa, 2025-01-13:
I had it87 compiled into the kernel, not as a module. So i recompiled the kernel with it87 as a module.
rebooted, modprobed it87 manually and it seems to work as it should. The module was not loaded automatically during boot.
By the way, sensors detect also found the chip.

Looks like it is failing to init the module during boot whenever that happens.

frankcrawford, 2025-01-14:
@VonKossa I haven't looked at compiling it into the kernel for sometime.  I will have a look when I have a chance, but it may be sometime, as I have a few other things I'm doing.

Thanks for following it up.

VonKossa, 2025-01-16:
No worries, take your time.
Thanks for responding.

VonKossa, 2025-01-19:
Fixed the title were is had written 6.8.12, i meant 6.12.8.

VonKossa, 2025-01-19:
Tested linux 6.6.67 and it could detect it87 when compiled into kernel.

VonKossa, 2025-03-21:
Unclear exactly when it started but problem remains in 6.12.19 and in 6.13 series.
- it is intermittent, sometimes it works and sometimes not
- when it doesn´t work this still show up in dmesg:
[    0.610975] it87: Found IT8728F chip at 0x290, revision 0
[    0.611060] it87: Beeping is supported

I actually thought it was a racing issue and something else prohibited the chip to be found sometimes but apparently not.

VonKossa, 2025-03-22:
it87 is compiled into the kernel and not as a module for me, so lsmod will not help here.

frankcrawford, 2025-03-23:
@VonKossa one item I just thought of here is that this module depends on the `hwmon_vid` module.  Are you compiling that in as well, or is it being loaded as a module, and if so, that could be a reason for the issues.

VonKossa, 2025-03-24:
Yes, it is also compiled into the kernel.

CONFIG_HWMON=y
CONFIG_HWMON_VID=y

tam3r, 2025-03-25:
Probably related to #43

VonKossa, 2025-03-26:
Thank you,
yes it might be related but that wouldn´t explain why it works intermittently. If it is a compression problem it should fail every time.

VonKossa, 2025-03-29:
nah, no cigarr, but i found the following in dmesg.
Don´t know why i didn´t see this before:

it87: Base address not set (chip IT8728F ioreg 0x2e), skipping

Do not know what it means, maybe you know?

frankcrawford, 2025-03-30:
@VonKossa okay, going back a bit, are you using the module distributed with the kernel, or are the one from this repo?

Secondly, what arguments are you specifying on loading?  Anything or just letting autoconfiguration do everything?

VonKossa, 2025-04-02:
Did some testing:
Compiled it87 as a module, modprobed and removed the module about 10-20 times, worked all the time.
Compiled it87 into the kernel, fails about 3 out of 4 times, the response in dmesg is either:
1. nothing at all mentioned in dmesg
2. it87: Base address not set (chip IT8728F ioreg 0x2e), skipping
3. Found IT8728F chip at 0x290, revision 0, but sensors show nothing

Compiled the kernel with debug chip option under hwmon but that didn´t show anything at all in dmesg.

So i am at a loss here, apparently something is seriously wrong when it87 is compiled into the kernel.

frankcrawford, 2025-04-03:
Yeah, although it may not be specifically `it87` that is the issue but the access to protected memory that early in the kernel boot process.

I suspect it is trying to read some memory block and failing, hence the issue.

I'm not sure I'll be able to troubleshoot this much.

VonKossa, 2025-04-03:
At the same time sensors_coretemp which i also use succeds to load everytime, so regardless it looks very specific to it87.
Anyhow, i agree, this can be caused for whatever reason, so troubleshooting this is not so easy, i will probably just load it as a module and in a year or two it will hopefully have auto fixed itself. It usually does. :)

frankcrawford, 2025-04-03:
Quite possibly, as they all read different memory and hardware blocks.  It is likely that something related to accessing the memory that the `it87` module uses is not yet initialised and needs to wait, but tracking down such an issue is going to be a very difficult and long process, if I even tried.

From your description it is not consistently getting the data from 0x290 either because something else is blocking it at the time or it is not yet populated, and that is probably because everything is still initialising.

The simplest fix is to run it as a module, which takes the loading and initial reads well away from any system initialisation issues.

And the reason why it probably started around 6.12.19 is that some other part of the kernel was reorganised causing the issues during initialisation of the module, not specifically the `it87` module itself.  There may be some lock or other step needed, but that is why I say tracking it down will be a long and involved process that I don't have time to do.

closed 2025-10-06 by frankcrawford