frankcrawford/it87 #43: modprobe it87 fails with dmesg message "decompression failed with status 6"

georgmittendorfer, 2025-01-18:
First of all thank you for your work on the kernel module, I can now see my fan speeds.

I had some issues installing it. I could not modprobe the file `it87.ko.xz` that was installed by make. dmesg showed the error message `decompression failed with status 6`.

I could fix this by replacing the `it87.ko.xz` module with the uncompressed `it87.ko` (+ depmod). Then I could successfully load the uncompressed module it with `modprobe it87 ignore_resource_conflict=1`.

System information:
```
6.12.9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.9-1 (2025-01-10) x86_64 GNU/Linux
Gigabyte Technology Co., Ltd. B650 AORUS ELITE AX/B650 AORUS ELITE AX, BIOS FA3a 02/06/2024
AMD Ryzen 9 7950X3D 16-Core Processor
```

sensors-detect reports the ITE chip ID `0x8689`. It is shown as `it8689-isa-0a40` with `sensors`.

frankcrawford, 2025-02-21:
@georgmittendorfer I will look at this a bit further, but this is probably not an issue with this module, but the build system in your distro.

frankcrawford, 2025-03-23:
Have you found any other details on what may be happening with the builds?

qubicmio, 2025-03-24:
No, I do not know why the incompatible compressed file was created. I posted the issue mainly for others so that they have a workaround in case they are affected. Feel free to close the issue.

tam3r, 2025-03-25:
I had this issue on Debian trixie with 6.12.19 and was able to load the module after adding "-C crc32" to xz call. Please see [this](https://github.com/lwfinger/rtw88/issues/189) for a possible fix.

frankcrawford, 2025-03-26:
@tam3r good catch.  Are you suggesting I should add something similar in the it87 Makefile?

frankcrawford, 2025-03-27:
Okay, I've added it and pushed it.  See how builds go now.

closed 2025-04-03 by frankcrawford