frankcrawford/it87 #46: Alternative for IT8613E with new linux kernels

john9631, 2025-02-10:
I was having difficulty using this solution with Linux Kernel 6.12.  Sometimes it would provide me with fan speeds, sometimes not.  So I did some additional research and andliuk at forums.truenas.com made this suggestion.

> I checked 6.12 kernel source, and it87 module is not updated to support IT8613E chipset. Nor it is on the latest kernel, so it is very unlikely it reaches production soon.
> 
> I found out the IT8622E chipset is similar to IT8613E, so as workaround the module can be loaded by forcing the chip id to 0x8622:
> 
> sudo modprobe it87 force_id=0x8622

He'd tried it on an Odroid H4+.  I used `dkms_clean` to remove the current modules.  Then I tried:

`sudo modprobe it87 force_id=0x8622`
followed by `sudo sensors-detect`
and `sensors` to check if the fan rpm was reported.  Success.

This was for a Beelink Mini S (Intel N100)  so it should work for other Intel N100 systems with Linux Kernel 6.12 or later.

john9631, 2025-02-10:
Note that to avoid using modprobe each time you reboot you can:
add the line `it87` at the end of /etc/modules-load.d/modules.conf
and add a file named `it87.conf` to `/etc/modprobe.d/ `and edit it to add the line `options it87 force_id=0x8622`

graysky2, 2025-02-14:
Are you sure you need the `force_id=` in the modprobe? Works for me without it,

frankcrawford, 2025-03-23:
@john9631 I think the reason you are seeing this behaviour is that you are loading the standard Linux kernel it87 module, not the enhanced one in the repo.

The standard one has the IT8622E defined, but not an IT8613E, so you would need a `force_id` option for it to work.

However, note that the are difference between the two, such as the number of temperatures supported.  But if it works for you that is fine as you may not be using them all.

closed 2025-03-23 by frankcrawford