frankcrawford/it87 #47: Gigabyte GA-B650M-D3HP working & lm_sensors fixed

JustinS-B, 2025-02-10:
Arch Linux, Kernel Version 6.13.1-arch1-1
Gigabyte Technology Co., Ltd. B650M D3HP motherboard

TLDR: lm_sensors all working on GA-B650M-D3HP with it87 running Arch Linux. Installed it87-dkms-git and edited lm_sensors.service

Original errors and problems
Running **sensors-detect** gives No, No, No except for finding an unknown chip with ID 0x8689 on a Gigabyte GA-B650M-D3HP motherboard running an AMD Ryzen 7 8700G cpu.

`Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689
Probing for Super-I/O at 0x4e/0x4f`

Trying to start **lm_sensors.service** gives the following error
`lm_sensors.service: Referenced but unset environment variable evaluates to an empty string: BUS_MODULES, HWMON_MODULES`

Fix: for Arch Linux, just install AUR/it87-dkms-git 

However, there are a couple of nice gotcha's that can cause you problems. 

Unless you have **/etc/sensors.d/GA-B650M-D3HP.conf** in place, you can't customise and of the sensor temperature settings. More importantly, unless you run **sensors -s** after each boot, none of your settings are applied. This is usually done by **lm_sensors.service**, but because the **sensors-detect** script doesn't like it87 chips, it failed to set the $BUS_MODULES and $HWMON_MODULES environmental variables and then didn't enable **lm_sensors.service**. However, it's easy to fix it all. Normally, all the lm_sensors scripts have a really clever way of enumerating and discovering the kernel modules that you need to read to get the temperature and fan data from the sensors. Alas, they just don't work for us because we have an it87 type chip. Thankfully, we know exactly how to load it & which specific options we need to use to get it all to work perfectly, so we can hard code that into the service file.

`sudo systemctl edit --full lm_sensors.service

[Unit]
Description=Initialize hardware monitoring sensors

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/sbin/modprobe it87 force_id=0x8689 ignore_resource_conflict=1
ExecStart=/usr/bin/sensors -s
ExecStop=/sbin/modprobe -r it87

[Install]
WantedBy=multi-user.target
`

Don't forget to do a **systemctl daemon-reload** and then enable it with **systemctl enable lm_sensors.service**

Copy this conf file into **/etc/sensors.d/GA-B650M-D3HP.conf** and reboot to check that everything works as expected

`# Gigabyte B650M D3HP GA-B650M-D3HP
#
# dmi: Board Manufacturer: Gigabyte Technology Co., Ltd.            
# dmi: Board Product Name: B650M D3HP
# dmi: BIOS Version: F22b
#
# 2025-02-07 JustinS-B
# Label mapping is mostly the same as for IT8686 on GA-B450M-DS3H and GA-B550M-DS3H
# https://github.com/frankcrawford/it87/blob/master/Sensors%20configs/GA-B450M-DS3H.conf
# But I added "nvme-pci-*", "gigabyte_wmi-virtual-*", and chip "acpitz-acpi-0" for completeness.

# Gigabyte B550M-DS3H
#
# dmi: Board Manufacturer: Gigabyte Technology Co., Ltd.            
# dmi: Board Product Name: B550<-DS3H
# dmi: BIOS Version: unknown
#
# 2023-09-24 DymOK93
# Label mapping is the same as for IT8686 on GA-B450M-DS3H

# Copied from 
# https://raw.githubusercontent.com/frankcrawford/it87/refs/heads/master/Sensors%20configs/GA-B550M-DS3H.conf

chip "it8689-*"
    label in0 "VCORE"
    label in1 "+3.3V"
        compute in1 @ * 1.65, @ / 1.65
    label in2 "+12.0V"
        compute in2 @ * 6, @ / 6
    label in3 "+5.0V"
        compute in3 @ * 2.5, @ / 2.5
    label in4 "VSOC"
    label in5 "VDDP"
    label in6 "DRAM"
    label in7 "3VSB"
    label in8 "VBAT"
    label fan1 "CPU Fan"
        set fan1_min  0
    label fan2 "SYS Fan 1"
    label fan3 "SYS Fan 2"
#    label fan4 "SYS Fan 3_PUMP"
#    label fan5 "CPU_OPT"
    label temp1 "System"
        set temp1_min  0.00
        set temp1_max  127.00
    label temp2 "Chipset"
        set temp2_min  0.00
        set temp2_max  127.00
    label temp3 "CPU"
        set temp3_min  0.00
        set temp3_max  127.00
    label temp4 "PCIe x16"
        set temp4_min  0.00
        set temp4_max  127.00
    label temp5 "VRM"
        set temp5_min  0.00
        set temp5_max  127.00
    label temp6 "VSoC"
        set temp6_min  0.00
        set temp6_max  127.00
    label intrusion0 "Intrusion"

# Note: The NAND Flash is always 8-10°C hotter than the Controller & the Composite readings
#       because NAND Flash runs hot...
chip "nvme-pci-*"

label temp1 "NVME_Composite"
  set temp1_min  0.5
  set temp1_max  127.50

label temp2 "NVME_Controller"
  set temp2_min  0.5
  set temp2_max  127.50

label temp3 "NVME_NAND_Flash"
  set temp3_min  0.5
  set temp3_max  127.50

# These are the same as temp sensors from chip "it8689-*" so you can safely ignore them if you want to
chip "gigabyte_wmi-virtual-*"

    label temp1 "System"
    label temp2 "Chipset"
    label temp3 "CPU"
    label temp4 "PCIe x16"
    label temp5 "VRM"
    label temp6 "VSoC"

#    ignore temp1
#    ignore temp2
#    ignore temp3
#    ignore temp4
#    ignore temp5
#    ignore temp6

chip "acpitz-acpi-0"
    label temp1 "CPU Die"

# For the record, "k10temp-pci-00c3" is a pseudo-temperature (see below),
# so it has been left "as is"
#
# info copied from
# https://www.kernel.org/doc/html/latest/hwmon/k10temp.html
#
# "Tctl is the processor temperature control value, used by the platform to 
# control cooling systems. Tctl is a non-physical temperature on an
# arbitrary scale measured in degrees. It does _not_ represent an actual
# physical temperature like die or case temperature. Instead, it specifies
# the processor temperature relative to the point at which the system must
# supply the maximum cooling for the processor's specified maximum case
# temperature and maximum thermal power dissipation."
#`

For the record, the **sensord.service** logging daemon has an unlisted dependency of extra/rrdtool without which it won't run.

frankcrawford, 2025-02-21:
@JustinS-B thanks for this.  I'll see what we can do about getting it into the distribution, although I will put this in the github repo.

frankcrawford, 2025-03-23:
@JustinS-B I've added your sensors conf file to the repo for others to be able to download, but I haven't done anything to get it into any of the upstream distributions.

Note, I've also left in it your definitions for "nvme-pci-*", "gigabyte_wmi-virtual-*", and chip "acpitz-acpi-0" even though it isn't related to the IT87 chipsets.

closed 2025-03-23 by frankcrawford

ChivuAndrei2003, 2025-04-08:
I wonder if this will work on my gigabyte b650m gaming x ax

nixternal, 2025-04-12:
> I wonder if this will work on my gigabyte b650m gaming x ax

I have the B650 Gaming X AX V2 and it works, only thing that doesn't work is setting the temp{1-6}_min & _max for any of the sensors, which isn't major, just annoying.

frankcrawford, 2025-04-14:
Yeah, that may be the case, I have noticed that sometime ITE/Gigabyte seem to randomly change registers between what look to be very similar chips, or even just drop the feature.

ChivuAndrei2003, 2025-04-17:
> @ChivuAndrei2003  probably will be close, although what chips are in the B650M Gaming X AX?
> 
> You can load it up and see how it goes, as it doesn't affect anything except the output of `sensors`, and you can easily remove or modify it if you find better solutions.

Nvm the problem got solved after I ran a script,rebooted and updated my system to the latest version.

Idk exactly which of these things I did fixed the problem🤣,but who cares as long as it works.