frankcrawford/it87 #49: Potential IT5570 Support?

benfertig, 2025-02-21:
I am running Proxmox VE 8.3 on a [Chatreey T9H](https://www.chatreey.com/drivers-manuals/t9h-mini-pc-drivers/) and was hoping to get PWM control over my fan, which I verified has 4 pins through physical examination (so, based on my understanding, this means that it should be controllable through PWM).

There are settings for the fan in the BIOS (Aptio Setup), but these don't seem to have any effect. The fan just turns on and off every 30 seconds or so regardless of what I put, although, to be fair, there are a LOT of settings, and I haven't tried ALL of them, nor do I know what they all mean. The main settings that I thought would make a difference were "Active trip point 0" and "Acive trip point 1".

Here is the Super I/O section of my ```sensors-detect``` output:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): yes
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x5570
```

I have gathered that ```0x5570``` represents the IT5570 chip. I have seen it suggested [here](https://github.com/lm-sensors/lm-sensors/issues/400#issuecomment-1304964842) that the IT5571 is equivalent to the IT8571, so I was thinking that a similar equivalency might apply between the IT5570 and IT8570. As a matter of fact, @frankcrawford, you seem to have been trying to help someone who was looking to get support added for the IT5571 chip in [this thread](https://github.com/frankcrawford/it87/issues/8), but doubted that it was possible.

Additionally, in that same thread, you mentioned that you looked at the spec sheet for the IT5570 and concluded that it does not have the right HW monitoring setup, so I am sorry if I am not contributing anything new here. I saw that you can get the spec sheet for the IT5570 [here](https://download.csdn.net/download/fxw12399/11855766), on the off chance that this could be a different document from the IT5570 spec sheet you were referring to. I only mention this because it took some digging to find, and I myself almost gave up trying to find it.

I saw that the "Download List" tab for the Chatreey T9H page that I linked includes some drivers, which appear to be Windows-specific (interestingly, there is also a download link for the BIOS but no actual download for the BIOS at the linked page, even though there is a BIOS download for the similar [Chatreey T9](https://www.chatreey.com/drivers-manuals/t9-model-mini-pc-drivers/)). I have not yet tested my T9H with Windows, and have therefore not tried installing those drivers to see if that would solve the fan issue.

Also, I assume that the manufacturer of this PC is AMI, seeing that the BIOS screen reads "Aptio Setup - AMI" at the top.

I'd be curious to know if you think there might be any chance at getting this driver to work with my chip!

frankcrawford, 2025-02-21:
@benfertig thanks fo the detailed write-up.  I will look into it, although it may take a month or two.

benfertig, 2025-02-21:
> [@benfertig](https://github.com/benfertig) thanks fo the detailed write-up. I will look into it, although it may take a month or two.

I really appreciate you already taking time to look into it! Let me know if there's ever something you need me to test on my end.

frankcrawford, 2025-03-23:
@benfertig I'm not sure I can do much here.  I can't get anything sensible from the link you supplied for the spec sheet, but also, the ITE chips you list don't seem to include hwmon capability, but are for other functions (not really sure what).

Is there anything else you can suggest to look up?

frankcrawford, 2025-03-23:
@benfertig from some information I have found about the IT5570 there may be something I can do.  I've found some specs and it looks to support voltage readings and fan control, although no temperature readings.

I can look to see how that corresponds to other chips currently supported and may be able to add something.  However, don't expect anything quickly.

benfertig, 2025-03-23:
@frankcrawford Looking forward to seeing what you find! Thank you for giving this the time and effort that you have. It's already encouraging just to see that there may still be hope in getting this to work.

I should mention that my Chatreey T9H supports CPU temperature readings. I believe it was through `coretemp`, but I can double-check if you need me to.

frankcrawford, 2025-03-24:
@benfertig there are multiple sensors and temperature detection within most modern motherboards.  As you noted one of the sensors would probably be `coretemp` which will list the temperatures within the CPU, and if that is all you need it will be fine.

It was mainly a note that the ITE hwmon chips measure temperatures at a number of points on the motherboard, not just the CPU, but this option won't be available with an IT5570.

Kabouik, 2025-04-09:
I stumbled upon this issue while looking at how to make an `nbfc-linux` config for the [Lunar](https://www.kickstarter.com/projects/1482776406/lunar-2-in-1-foldable-keyboard-and-mini-pc-for-all-your-needs) keyboard miniPC, which also uses an ITE5570 according to `superiotool`. I don't know yet how I could the progress made here for finally being able to control the fan in my sytem, but I wanted to chime in to say that I'm super interested too ad there is a community of other users who are also interested in being able to control fan speed in Linux for this hardware.

I posted some other information in the other issue I referenced just above, including a `superiotool --dump`, in case it might be useful

Kabouik, 2025-04-09:
The datasheet [IT5570_A_V0.3.1_U.pdf](https://github.com/user-attachments/files/19673123/IT5570_A_V0.3.1_U.pdf) looks complete and may be of help. I uploaded the file to Github out of security.

We've found that pages 418 onwards are the PWM section and may be interesting to you @frankcrawford. It includes reading fan tachs and setting PWM duty cycles. Page 420 gives the formulas for going from 0% to 100%. There is also a programming flow chat for PWM output. The manufacturer recommends disabling the PWM clock (PWM inhibit mode = fan off) while changing PWM registers.

I don't know if my machine's firmware overrides whatever we would set in the registers, but at least I know the manufacturer has made a Windows app that *can* take control of the fan speed, so there's hope the firmware stays out of the way (or maybe there's another register to flip to disable any override?). We'll try to reach the manufacturer, see if we can get information as to how their Windows app does it.

Someone also mentioned that "EC registers you'll be interested in are 02h-09h for PWM duty cycle control, 1eh/1fh for lsb/msb tach for fan1, and 20h/21h for lsb/msb tach for fan2."

Thanks to members `Tsunami` from CoolerControl Discord group and `charlie` from the Lunar Discord group for their help putting together this.

Kabouik, 2025-04-10:
I realized I have another computer model using IT5570, the GPD Pocket 3. It has a hardware key to support two distinct curves: normal, and quiet.

Writing to `0x70` did change the fan speed on it (temporarily), but I couldn't find which register to read from. Weirdly, on the Lunar (my other computer model using IT5570), writing to `0x70` didn't change the fan speed, even though it was the register I identified with `ec_probe` too. I guess there might be different implementations?

frankcrawford, 2025-04-14:
I can see you are keen on this, and that is good.  I will see what I can do in the near future about adding it in a branch.

The link you gave is probably for the same chip, but a different feature.  the ITx chipset managest a lot of low level functions, from keyboards, to floppies, and other older components.  We only care about the hardware monitoring part here, but other parts of the kernel do make use of some other features.

Sometimes they don't need to know details of the chip itself, as the API is very standard.  Unfortunately for hwmon that isn't the case.

Kabouik, 2025-04-30:
Thanks for your reply, @frankcrawford! 

Just in case it's useful (I'm sorry, I'm not skilled to filter what is relevant or not here), someone also collected more information on the chip here last week: https://github.com/nbfc-linux/nbfc-linux/issues/188#issuecomment-2817427909