frankcrawford/it87 #50: Support Gigabyte X399 Designare EX

nicholascw, 2025-03-10:
Hi, the board_name is `X399 DESIGNARE EX-CF` and it is a 8686E + 8792/8795E configuration which needs an update in the dmi table. I tried to run a patched version on my side and it works. Also tried `ignore_resource_conflict=1` which would lead to a system freeze while running `sensors-detect`.

frankcrawford, 2025-03-17:
@nicholascw is what you list for the board_name as returned by the the file in /sys?

Also, `sensors_detect` is doesn't really know about his module, so breaking with it is not a surprise.  It does low-level stuff that may break things.

Anyway, I'll add it shortly.

nicholascw, 2025-03-21:
Yes that's grabbed from `cat /sys/class/dmi/id/board_name`

frankcrawford, 2025-03-23:
Anyway, I've added it into the DMI table, lets see how it goes.

closed 2025-03-23 by frankcrawford