frankcrawford/it87 #51: Support: X870 AORUS ELITE WIFI7 ICE

marrobHD, 2025-03-19:
Hi,

the board name is `X870 AORUS ELITE WIFI7 ICE`.

sensors-detect:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8696
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8883
    (logical device 4 has address 0x290, could be sensors)

```

isadump:
```
sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 96 00 00 2e 20 b3 00 00 00 80 30 88 40 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 10 0a 00 0a 20 00 00 00 00 00 00 00 00 00 00 
70: 00 00 20 00 00 17 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 c0 00 00 00 00 
c0: 01 00 00 40 00 ff ff ff 01 00 00 00 00 60 00 00 
d0: b5 00 08 4a 00 fd ff ff 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 50 00 20 12 49 00 00 0a 7c 
f0: 00 00 00 00 00 00 0d 00 00 d0 49 60 ff a5 00 0f 

sudo isadump 0x295 0x296
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
10: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
20: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
30: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
40: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
50: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
60: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
70: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
80: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
90: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
a0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
b0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
c0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

frankcrawford, 2025-03-22:
@marrobHD I have a PR to add the it8696 shortly and so you should be able to use it soon.

Do you know if this board has 1 or 2 ITE hwmon chips?  The dump for 0x295 to 0x296 looks to have returned no results, which could be for a few reasons, including that `sensors-detect` has locked up the chip, which is a known issue.

Anyway, see what happens when I add the first chip shortly.

marrobHD, 2025-03-22:
Hmm I'm not 100% sure. In the manual there's a picture which features just one super I/o ite Chip. 

Anand thanks :)

frankcrawford, 2025-03-23:
Yeah, the manual is usually pretty poor on the chips, but I guess if `sensors-detect` is only picking up one, then it is likely to be just the one.

Anyway, I've added it in, lets see how it goes.

closed 2025-03-23 by frankcrawford

marrobHD, 2025-03-24:
Seems to work:
```
it8696-isa-0a40
Adapter: ISA adapter
in0:           1.15 V  (min =  +0.00 V, max =  +3.06 V)
in1:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:           1.98 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.08 V  (min =  +0.00 V, max =  +3.06 V)
in5:           1.15 V  (min =  +0.00 V, max =  +3.06 V)  ALARM
in6:           1.14 V  (min =  +0.00 V, max =  +3.06 V)  ALARM
3VSB:          3.31 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.00 V  
+3.3V:         3.07 V  
fan1:        1138 RPM  (min =    0 RPM)
fan2:         436 RPM  (min =    0 RPM)
fan3:         425 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:        2528 RPM  (min =    0 RPM)
temp1:        +22.0°C  (low  = +127.0°C, high = +127.0°C)
temp2:        +34.0°C  (low  = +127.0°C, high = +127.0°C)
temp3:        +31.0°C  (low  = +127.0°C, high = +127.0°C)
temp4:        +27.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +24.0°C  (low  =  +0.0°C, high = -125.0°C)
temp6:        -55.0°C  (low  = +127.0°C, high = +127.0°C)
intrusion0:  ALARM

it87952-isa-0a60
Adapter: ISA adapter
in0:           1.80 V  (min =  +0.00 V, max =  +2.81 V)
in1:         385.00 mV (min =  +0.00 V, max =  +2.81 V)
in2:           1.84 V  (min =  +0.00 V, max =  +2.81 V)
+3.3V:       242.00 mV (min =  +0.00 V, max =  +5.61 V)
in4:           1.80 V  (min =  +0.00 V, max =  +2.81 V)
in5:           1.06 V  (min =  +0.00 V, max =  +2.81 V)
in6:           1.41 V  (min =  +0.00 V, max =  +2.81 V)
3VSB:          3.39 V  (min =  +0.00 V, max =  +5.61 V)
Vbat:          2.97 V  
fan1:        1011 RPM  (min =    0 RPM)
fan2:        1013 RPM  (min =    0 RPM)
fan3:         739 RPM  (min =    0 RPM)
temp1:        +25.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +25.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```