frankcrawford/it87 #52: IT8613E Fan 5 Speed Sensor Not Detected

hboyd2003, 2025-04-04:
I am running a Minisforum BD790i/BD795i SE which has the ITE IT8613E (manually confirmed) with the latest version of the driver the speed sensor for Fan 5 does not show up.

After modifying the driver to stop skipping the fan 5 speed sensor everything worked fine.

<details><summary>Output from isadump</summary>
<p>

```
redacted@redacted:/redacted/it87$ sudo isadump -y -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7 
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
20: 86 13 0c 40 00 00 fb 02 04 01 01 48 43 00 00 00 
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 10 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 01 20 38 00 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: ff ff ff ff ff ff ff ff 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 c0 03 00 00 
c0: 01 08 00 44 00 00 00 00 00 08 00 04 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 b0 44 00 00 00 00 
e0: 00 00 00 00 00 00 00 10 00 1d d1 dc 00 00 40 04 
f0: 00 00 00 00 00 00 0e 00 22 08 00 00 7c 00 00 00 
redacted@redacted:/redacted/it87$ sudo isadump -y 0xa35 0xa36
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 19 90 00 00 00 00 00 00 00 c0 48 00 30 00 6a 5e 
10: ff ff ff 60 c0 00 00 08 00 01 01 ff ff ff 10 08 
20: 3f 62 b2 b2 b9 a2 a4 94 91 1c 1d 27 00 1c 1c 97 
30: ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 ff 00 
40: 7f 80 7f 80 7f 80 7f 80 ad 6a d4 00 5f 40 74 12 
50: ff 18 7f 7f 7f 40 00 00 90 00 00 ff 60 00 00 00 
60: 7f 7f 7f 80 00 00 0f 80 7f 7f 7f 87 00 00 0f 87 
70: 21 1e 50 ff 10 02 0f ff 00 19 5e ff 14 02 0f ff 
80: 82 01 a1 00 ff ff ff ff 00 00 00 00 00 fb 00 bd 
90: 00 00 00 00 00 00 00 ff 42 99 00 00 01 8d d2 27 
a0: 00 28 5c ff 10 02 0f ff 7f 7f 7f 80 00 00 0f 80 
b0: da ab ff ff 7f 80 7f 80 a3 e8 06 0d f0 14 c9 05 
c0: 00 00 00 10 91 01 10 82 ff ff ff ff ff ff ff ff 
d0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 01 
e0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
f0: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

</p>
</details> 

Let me know what other info you may need.

frankcrawford, 2025-06-26:
I'll have to see what the specs for this chip are.

woinaT, 2025-09-08:
My wife is Chinese :-) and ChatGPT produced a convincing story about a new hardware design project incorporating the chip :-(.

frankcrawford, 2025-09-08:
Well, related, did she get a contact address I can try?

woinaT, 2025-09-08:
I have forwarded you the mail

frankcrawford, 2025-10-06:
@woinaT I've now updated and pushed this fix.  It is in the latest version on the repo.

closed 2025-10-06 by frankcrawford