frankcrawford/it87 #59: 127c temp2

SteavenGamerYT, 2025-06-22:
it8665-isa-0290
Adapter: ISA adapter
in0:           1.00 V  (min =  +0.83 V, max =  +2.32 V)
in1:           2.51 V  (min =  +2.07 V, max =  +0.68 V)  ALARM
in2:           2.00 V  (min =  +1.37 V, max =  +1.29 V)  ALARM
in3:           2.00 V  (min =  +2.26 V, max =  +2.77 V)  ALARM
in4:          33.00 mV (min =  +2.73 V, max =  +1.82 V)  ALARM
in5:          33.00 mV (min =  +0.67 V, max =  +2.05 V)  ALARM
in6:          33.00 mV (min =  +1.92 V, max =  +2.74 V)  ALARM
3VSB:          3.33 V  (min =  +1.16 V, max =  +1.00 V)  ALARM
Vbat:          3.25 V  
+3.3V:         3.36 V  
fan1:        1516 RPM  (min =   12 RPM)
fan2:           0 RPM  (min =  174 RPM)
fan3:        7031 RPM  (min =   25 RPM)
fan4:           0 RPM  (min =   -1 RPM)
fan6:           0 RPM  (min =   -1 RPM)
temp1:        +51.0°C  (low  = -89.0°C, high = -20.0°C)  ALARM
temp2:       +127.0°C  (low  = +31.0°C, high = -65.0°C)  ALARM  sensor = thermal diode
temp3:        +31.0°C  (low  = -102.0°C, high = -74.0°C)  ALARM  sensor = thermistor
temp4:        +31.0°C  (low  = -115.0°C, high = -92.0°C)  ALARM  sensor = thermistor
temp5:        +31.0°C  (low  =  -6.0°C, high = -16.0°C)  ALARM  sensor = thermistor
temp6:        +31.0°C  (low  = -70.0°C, high = -77.0°C)  ALARM  sensor = thermistor
intrusion0:  ALARM

temp2:       +127.0°C  (low  = +31.0°C, high = -65.0°C)  ALARM  sensor = thermal diode

this has to bug right? 127c?
i am using nixos 25.05

SteavenGamerYT, 2025-06-27:
[image] you sure?
they all show correct values most of the time

SteavenGamerYT, 2025-06-27:
this what bios has
[image]

this what linux says most of the time
this is after reboot
[image]

SteavenGamerYT, 2025-06-27:
i think that temp1 is cpu
temp2 is motherboard
temp3 to temp6 is not real as you said

frankcrawford, 2025-06-27:
What motherboard do you actually have as well?  The other ones may be valid on the chip, but either not used by you or not even connected to anything on that M/B.

So that really means that they should be ignored for that particular M/B, which is actually an lm_sensors setting.

frankcrawford, 2025-06-27:
However, you are probably right that there is something going on with temp2 which needs to be looked at, but also need to see what the M/B is really doing.

SteavenGamerYT, 2025-08-05:
[dmesg-logs.txt](https://github.com/user-attachments/files/21600006/dmesg-logs.txt)
[journalctl-logs.txt](https://github.com/user-attachments/files/21600007/journalctl-logs.txt)

frankcrawford, 2025-10-06:
Ahh, that is good.  It means it was measuring right, but you need that sensor file to convert it to the correct units.

SteavenGamerYT, 2025-11-19:
[journalctl.log](https://github.com/user-attachments/files/23616268/journalctl.log)
[dmesg.log](https://github.com/user-attachments/files/23616267/dmesg.log)

https://github.com/user-attachments/assets/64ca79eb-fc2c-4874-9551-77d000430c28