frankcrawford/it87 #60: Add Support for B850M Aorus Elite Ice 0x8696

blipk, 2025-06-22:
Seems to be the same chips as this but it's not working for me: https://github.com/frankcrawford/it87/issues/51

I'm using this dkms package: https://aur.archlinux.org/packages/it87-dkms-git

Dumps requested from readme:

```
>sudo sensors-detect

# sensors-detect version 3.6.2
# System: Gigabyte Technology Co., Ltd. B850M AORUS ELITE WIFI6E ICE [Default string-CF-WCP-ADO]
# Kernel: 6.15.2-arch1-1 x86_64
# Processor: AMD Ryzen 7 9800X3D 8-Core Processor (26/68/0)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): yes
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
AMD Family 19h thermal sensors...                           No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): yes
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8696
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): yes
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): yes
Probing for `National Semiconductor LM78' at 0x290...       No
[... 372 lines cut ...]
intrusion0:  ALARM

amdgpu-pci-1000
Adapter: PCI adapter
vddgfx:        1.17 V  
vddnb:         1.19 V  
edge:         +37.0°C  
PPT:           9.00 mW 
sclk:         600 MHz 

gigabyte_wmi-virtual-0
Adapter: Virtual device
temp1:        +26.0°C  
temp2:        +41.0°C  
temp3:        +40.0°C  
temp4:        +29.0°C  
temp5:        +33.0°C  

nvme-pci-0f00
Adapter: PCI adapter
Composite:    +33.9°C  (low  =  -0.1°C, high = +99.8°C)
                       (crit = +109.8°C)
```

If there's anything else that would help, let me know.

blipk, 2025-06-22:
Voltages in BIOS

[image]
[image]
[image]

blipk, 2025-06-27:
Thanks @frankcrawford 

Tag my name if you need any more info

blipk, 2025-06-30:
Hi. I can confirm the chipset is ITE 5711 as that is the driver provided from the gigabyte website.

blipk, 2025-07-01:
Even more specs here: https://github.com/developersu/argbColors/blob/master/SPECIFICATION.md

Willing to donate/contribute to help get this implemented - do you have an option for that? @frankcrawford

blipk, 2025-07-01:
Ok maybe the IT 5711 chip is unrelated to fans - It's what they supply here under utilities for the ARGB controller: https://www.gigabyte.com/Motherboard/B850M-AORUS-ELITE-WIFI6E-ICE-rev-1x/support#support-dl

I photographed the chip on my board though and it says IT8696E

[image]

blipk, 2025-07-01:
LibreHardwareMonitor seems to support this chip but is windows only, might be useful to see their implementation: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor

frankcrawford, 2025-07-01:
The IT8689E is already supported, although if it is not detected on your motherboard, we need to find out why.

Firstly, reboot your maching and then load the it87 module without a "force_id" option and see if it is detected.  Don't run `sensors-detect`.  Check to see what comes up in `dmesg` about the it87 module.

Following that we can see what is happening on autoload, it doesn't work.

BTW, you are likely to find that you cannot set the fan settings, although you can view them.  That is a different issue that is a know problem.

Now as for the ITE 5711, given you are looking at ARGB and it is referenced on the LiquidCtl website, it probably a USB chip not a hwmon chip.

blipk, 2025-07-02:
Hey @frankcrawford 

It's actually `IT8696E` but you wrote `IT8689E` is supported - not sure if typo.

Remember I'm using this package which uses dkms but is built from your source here: https://aur.archlinux.org/packages/it87-dkms-git

There's no mention of it in dmesg at all:

<details>
<summary>dmesg output</summary>

```
[    0.000000] Linux version 6.15.2-arch1-1 (linux@archlinux) (gcc (GCC) 15.1.1 20250425, GNU ld (GNU Binutils) 2.44.0) #1 SMP PREEMPT_DYNAMIC Tue, 10 Jun 2025 21:32:33 +0000
[    0.000000] Command line: initrd=\9b8c16666fc744cc8479fdec0ca7f9bc\6.15.2-arch1-1\initrd nvme_load=YES nowatchdog rw rd.luks.uuid=0e30d180-b75f-4912-808a-f02e53026d0f root=/dev/mapper/luks-0e30d180-b75f-4912-808a-f02e53026d0f systemd.machine_id=9b8c16666fc744cc8479fdec0ca7f9bc
[    0.000000] x86/split lock detection: #DB: warning on user-space bus_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009afefff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009aff000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a21dfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a21e000-0x000000000affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000b000000-0x000000000b020fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000b021000-0x0000000088023fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000088024000-0x000000008e023fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008e024000-0x000000008e27efff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008e27f000-0x000000009027efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000009027f000-0x00000000983fefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000983ff000-0x0000000099ff6fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000099ff7000-0x0000000099ffbfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000099ffc000-0x0000000099ffffff] usable
[    0.000000] BIOS-e820: [mem 0x000000009a000000-0x000000009bffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000009d7f3000-0x000000009fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f7000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000083de7ffff] usable
[    0.000000] BIOS-e820: [mem 0x000000083eec0000-0x00000008801fffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000fd00000000-0x000000ffffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] efi: EFI v2.9 by American Megatrends
[    0.000000] efi: ACPI=0x90267000 ACPI 2.0=0x90267014 TPMFinalLog=0x90233000 SMBIOS=0x978d3000 SMBIOS 3.0=0x978d2000 MEMATTR=0x84119318 ESRT=0x8405de98 RNG=0x8e09bf18 INITRD=0x7edfa318 TPMEventLog=0x8e097018 
[    0.000000] random: crng init done
[    0.000000] efi: Remove mem255: MMIO range=[0xe0000000-0xefffffff] (256MB) from e820 map
[    0.000000] e820: remove [mem 0xe0000000-0xefffffff] reserved
[    0.000000] efi: Remove mem256: MMIO range=[0xf7000000-0xfedfffff] (126MB) from e820 map
[    0.000000] e820: remove [mem 0xf7000000-0xfedfffff] reserved
[    0.000000] efi: Not removing mem257: MMIO range=[0xfee00000-0xfee00fff] (4KB) from e820 map
[    0.000000] efi: Remove mem258: MMIO range=[0xfee01000-0xffffffff] (17MB) from e820 map
[    0.000000] e820: remove [mem 0xfee01000-0xffffffff] reserved
[    0.000000] efi: Remove mem260: MMIO range=[0x860000000-0x8801fffff] (514MB) from e820 map
[    0.000000] e820: remove [mem 0x860000000-0x8801fffff] reserved
[    0.000000] SMBIOS 3.7.0 present.
[    0.000000] DMI: Gigabyte Technology Co., Ltd. B850M AORUS ELITE WIFI6E ICE/B850M AORUS ELITE WIFI6E ICE, BIOS F4 05/23/2025
[    0.000000] DMI: Memory slots populated: 2/4
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 4691.279 MHz processor
[    0.000110] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000112] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000116] last_pfn = 0x83de80 max_arch_pfn = 0x400000000
[    0.000122] MTRR map: 5 entries (3 fixed + 2 variable; max 20), built from 9 variable MTRRs
[    0.000123] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000281] e820: update [mem 0xa0000000-0xffffffff] usable ==> reserved
[    0.000286] last_pfn = 0x9a000 max_arch_pfn = 0x400000000
[    0.003221] esrt: Reserving ESRT space from 0x000000008405de98 to 0x000000008405ded0.
[    0.003227] e820: update [mem 0x8405d000-0x8405dfff] usable ==> reserved
[    0.003245] Using GB pages for direct mapping
[    0.003442] Secure boot disabled
[    0.003442] RAMDISK: [mem 0x67425000-0x72949fff]
[... 1473 lines cut ...]
[ 4466.273526] wlan0: associated
[ 5046.655736] wlan0: Connection to AP 84:d8:1b:25:4b:d5 lost
[ 5050.661083] wlan0: authenticate with 84:d8:1b:25:4b:d5 (local address=60:ff:9e:4f:2a:84)
[ 5050.661088] wlan0: send auth to 84:d8:1b:25:4b:d5 (try 1/3)
[ 5050.663510] wlan0: authenticated
[ 5050.664669] wlan0: associate with 84:d8:1b:25:4b:d5 (try 1/3)
[ 5050.668577] wlan0: RX AssocResp from 84:d8:1b:25:4b:d5 (capab=0x431 status=0 aid=1)
[ 5050.771760] wlan0: associated
[... cut ...]
[ 5177.526800] wlan0: send auth to 84:d8:1b:25:4b:d5 (try 1/3)
[ 5177.529221] wlan0: authenticated
[ 5177.529419] wlan0: associate with 84:d8:1b:25:4b:d5 (try 1/3)
[ 5177.532891] wlan0: RX AssocResp from 84:d8:1b:25:4b:d5 (capab=0x431 status=0 aid=1)
[ 5177.642507] wlan0: associated
```
</details>

Nothing in journalctl either but dkms shows the module as installed:
```
❯ dkms status
it87/287.4bff981, 6.15.2-arch1-1, x86_64: installed (Original modules exist)
```

Any tips on what else I should check?

frankcrawford, 2025-07-02:
Sorry, the earlier one is a typo, I don't know where it came from.

Anyway, you probably need to create a file `/etc/modprobe.d/it87.conf` with 

```
# Local IT87 sensor options - no longer required
options it87 ignore_resource_conflict=1
```
That should make it good to load on boot.

blipk, 2025-07-02:
Thanks @frankcrawford  but unfortunately that didn't work.

I've posted over on the AUR to see if they can help out.

paquiteau, 2025-07-15:
Hello there, I recently got a Gigabyte B850M D3HP motherboard, and had the same troubleshooting steps as @blipk, and after applying the same fixes, I got similar outputs:

```

it8696-isa-0a40
Adapter: ISA adapter
in0:         480.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           1.99 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:         912.00 mV (min =  +0.00 V, max =  +3.06 V)
in5:           1.12 V  (min =  +0.00 V, max =  +3.06 V)  ALARM
in6:           1.13 V  (min =  +0.00 V, max =  +3.06 V)  ALARM
3VSB:          3.31 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.07 V  
+3.3V:         3.07 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:        1001 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan5:        1022 RPM  (min =    0 RPM)
temp1:        +32.0°C  (low  = +127.0°C, high = +127.0°C)
temp2:        +54.0°C  (low  = +127.0°C, high = +127.0°C)
temp3:        +40.0°C  (low  = +127.0°C, high = +127.0°C)
temp4:        +35.0°C  (low  = +127.0°C, high = +127.0°C)
temp5:        +46.0°C  (low  =  +0.0°C, high = -125.0°C)
temp6:         +0.0°C  (low  =  +0.0°C, high = +127.0°C)
intrusion0:  ALARM
```

Thank you all for working with open garage doors. Is there any chance to tune the fan speed directly from the OS (without going back to bios) ?  I can test further things if needed.

frankcrawford, 2025-07-16:
@paquiteau unfortunately at present, no we can't tune the fans directly, as it looks like Gigabyte has added another chip in somewhere that does this.  See the discussions in #11.

I am planning to do some testing on this later this year, but until then I won't have much time to play with it.

closed 2025-07-23 by blipk